coreos/bootupd #1: Fixup the fwupd bit?

hughsie, 2020-06-26:
> fwupd comes as a UEFI binary today, so bootupd would actually take care of updating fwupd itself.

At the moment to schedule an update fwupd

* mounts the ESP if not already mounted
* copies the `/usr/libexec/fwupd/efi/fwupdx64.efi.signed` binary to `EFI/fedora/fwupdx64.efi`
* copies the firmware from the archive to `EFI/fedora/fw/fwupd-{guid}.cap`
* futzes with the BootOnce EFI key
* unmounts the ESP if it wasn't mounted before we started

...so we don't actually care if `fwupdx64.efi` is old or insecure or whatever, as it's going to be overwritten on next firmware update anyway. We could even mount->rm  `EFI/fedora/fwupdx64.efi` -> unmount on next startup if that helps, although I'm not sure it's a helpful thing to do.

We did used to install into the ESP fro the rpm %install phase, but various distros don't automount the ESP (perhaps sensibly) and so we stopped doing that.

If I've massively misunderstood the problem, apologies.

cgwalters, 2020-06-26:
Thanks for filing this!  I think we'll need to debate a bit who owns the updates here.  To start it's probably ok if bootupd defers to fwupd to update itself.

hughsie, 2020-06-26:
> I think we'll need to debate a bit who owns the updates here

If there's an API to use I'd be fine adding code to fwupd to say "_put $THAT$ file on the ESP_" and just let something else deal with mounting/unmounting/copying etc.

cgwalters, 2020-06-26:
Right.  I think there's a lot more to "have fwupd defer updates to bootupd" though like how versioning is handled and whether we want to support updating fwupd separately from other things etc.

Conceptually fwupd is a lot more like an "EFI app" which is quite distinct from the OS bootloaders I think.

I just pushed a change to bootupd which drops the concept that it has exclusive control over `/boot/efi`, instead it will only adjust the files it knows about.  So for now I'll move forward operating under the idea that bootupd needs to not break what fwupd is doing today, and then we can iterate towards having the cooperate once bootupd is stable (and hence fwupd could in theory depend on it, then we have to ensure that everywhere fwupd is adopted bootupd is too etc).

hughsie, 2020-06-26:
> Conceptually fwupd is a lot more like an "EFI app" 

Agree.

> instead it will only adjust the files it knows about

Great, thanks.

> and hence fwupd could in theory depend on it

I think a soft-depend is fine, we try to keep the list of deps as small as possible. Yell when you've got API you want us to use.

closed 2020-09-03 by ashcrow