coreos/bootupd #43: implement auto-exit on idel

cgwalters, 2020-09-28:
I am pretty sure at one point we did the "auto-exit on idle" pattern but it doesn't look like we are today.  We should just to conserve resources.

Let's depend on tokio and add an idle timeout in the daemon.

lucab, 2020-10-01:
I think the ~daemon~ devil is in the details here, it looks like "depend on tokio" is a much larger task which first requires making sure that nothing in the daemon logic blocks the executor and things can be gracefully canceled, no? In particular, the `accept` [step](https://github.com/coreos/bootupd/blob/bfd47c93dbff810722ef333151ed17a4fb267cdd/src/daemon/mod.rs#L64) indefinitely blocks right now and can't be retired (and every IPC calls from there down too).

Perhaps a shorter-term goal with an higher immediate benefit is to process a single client and then exit. That would induce the same start-latency upfront, and make sure that we don't mistakenly start cobbling things that assume an always running service (e.g pull-based metrics).

cgwalters, 2020-10-01:
> Perhaps a shorter-term goal with an higher immediate benefit is to process a single client and then exit.

Yep, that seems simpler indeed.

cgwalters, 2020-10-01:
Well I tried
```
diff --git a/src/cli/bootupd.rs b/src/cli/bootupd.rs
index 377bf65..4a3834d 100644
--- a/src/cli/bootupd.rs
+++ b/src/cli/bootupd.rs
@@ -57,7 +57,7 @@ impl DCommand {
     /// Run CLI application.
     pub fn run(self) -> Result<()> {
         match self.cmd {
-            DVerb::Daemon => crate::daemon::run_coreloop(),
+            DVerb::Daemon => crate::daemon::run(),
             DVerb::Install(opts) => Self::run_install(opts),
             DVerb::GenerateUpdateMetadata(opts) => Self::run_generate_meta(opts),
         }
diff --git a/src/daemon/mod.rs b/src/daemon/mod.rs
index f1ed85d..f5209b4 100644
--- a/src/daemon/mod.rs
+++ b/src/daemon/mod.rs
@@ -5,26 +5,29 @@ use anyhow::{bail, Context, Result};
 use nix::sys::socket as nixsocket;
 use std::os::unix::io::RawFd;
 
-/// Run daemon core-logic loop, endlessly.
-pub fn run_coreloop() -> Result<()> {
+/// Accept a single client and then exit; we don't want to
+/// persistently run as a daemon.  The systemd unit is mostly
+/// and implementation detail - it lets us use things like
+/// systemd's built in sandboxing (ProtectHome=yes) etc. and also
+/// ensures that only a single bootupd instance is running at
+/// a time (i.e. don't support concurrent updates).
+pub fn run() -> Result<()> {
     let srvsock_fd = systemd_activation().context("systemd service activation error")?;
 
-    loop {
-        // Accept an incoming client.
-        let client = match accept_authenticate_client(srvsock_fd) {
-            Ok(auth_client) => auth_client,
-            Err(e) => {
-                log::error!("failed to authenticate client: {}", e);
-                continue;
-            }
-        };
-
-        // Process all requests from this client.
-        if let Err(e) = process_client_requests(client) {
-            log::error!("failed to process request from client: {}", e);
-            continue;
+    // Accept an incoming client.
+    let client = match accept_authenticate_client(srvsock_fd) {
+        Ok(auth_client) => auth_client,
+        Err(e) => {
+            log::error!("failed to authenticate client: {}", e);
+            return Ok(());
         }
+    };
+
+    // Process all requests from this client.
+    if let Err(e) = process_client_requests(client) {
+        log::error!("failed to process request from client: {}", e);
     }
+    Ok(())
 }
 
 /// Perform initialization steps required by systemd service activation.
diff --git a/tests/kola/test-bootupd b/tests/kola/test-bootupd
index 964f243..e9e38fd 100755
--- a/tests/kola/test-bootupd
+++ b/tests/kola/test-bootupd
@@ -48,6 +48,10 @@ assert_file_has_content_literal out.txt '  Installed: grub2-efi-x64-'
 assert_file_has_content_literal out.txt 'Update: At latest version'
 ok status
 
+# Validate we auto-exited
+systemctl show -p ActiveState bootupd > out.txt
+assert_file_has_content_literal out.txt 'ActiveState=inactive'
+
 bootupctl validate | tee out.txt
 ok validate
 
```

but
```
Oct 01 19:00:39 localhost.localdomain systemd[1]: bootupd.service: Start request repeated too quickly.
Oct 01 19:00:39 localhost.localdomain systemd[1]: bootupd.service: Failed with result 'start-limit-hit'.
Oct 01 19:00:39 localhost.localdomain systemd[1]: Failed to start bootloader update daemon.
```

So yeah...probably need to do exit-on-idle indeed.

closed 2020-10-02 by openshift-merge-robot