coreos/bootupd #132: Support updating multiple EFIs in mirrored setups

jlebon, 2021-01-06:
Follow-up to https://github.com/coreos/bootupd/pull/127 (specifically https://github.com/coreos/bootupd/pull/127#discussion_r546995248). We need to update the EFI in all the devices matching the appropriate GUID, not just the one which wins the `/dev/disk/by-partlabel/EFI-SYSTEM` race.

bgilbert, 2021-01-07:
One small correction: in the mirrored boot disk case, no partitions are labeled `EFI-SYSTEM`.

jlebon, 2021-12-02:
Note that just dumbly doing a type GUID match will not work for things like multipath. We solved this in `rdcore` by looking for ESPs only on the same devices which back the boot partition: https://github.com/coreos/coreos-installer/blob/78e2a7a4bf8d9ddb90b5e4fb5729a2550999ad11/src/blockdev.rs#L880-L945

Would probably make sense for bootupd to use the same heuristic (or if it finds a better heuristic to use, update coreos-installer to match).

travier, 2024-06-27:
Notes for implementing this for Fedora CoreOS:
- https://github.com/coreos/fedora-coreos-tracker/issues/1485#issuecomment-1533350911
- https://github.com/coreos/fedora-coreos-tracker/issues/1752#issuecomment-2194950462

But unfortunately we might have to add more heuristics to support Anaconda installations.

travier, 2024-09-02:
As far as I could see when tried various RAID setups in Anaconda (https://docs.fedoraproject.org/en-US/fedora/f36/install-guide/install/Installing_Using_Anaconda/#sect-installation-gui-manual-partitioning-swraid), they only create a single EFI system partition.

So Fedora CoreOS is the only case with one per disk.

See also: https://gitlab.com/fedora/ostree/sig/-/issues/1#note_2086744541

HuijingHei, 2024-09-26:
Test failed with multiple EFIs in mirrored setups (using latest bootupd in copr build):

```
$ cosa run --add-disk 10G -B ../ign/mirror.bu --qemu-firmware uefi -m 4096
[root@cosa-devsh ~]# mount -o remount,rw /boot
[root@cosa-devsh ~]# mv /boot/bootupd-state.json .
[root@cosa-devsh ~]# bootupctl adopt-and-update
Running as unit: bootupd.service
lsblk: : not a block device
error: Failed to run lsblk
```

mirror.bu is refered to [doc](https://docs.fedoraproject.org/en-US/fedora-coreos/storage/#_reconfiguring_the_root_filesystem)
```
variant: fcos
version: 1.5.0
boot_device:
  mirror:
    devices:
      - /dev/vda
      - /dev/vdb
```

closed 2025-06-16 by dustymabe