coreos/bootupd #185: Readme does not explain whether/how vendors need to support this

rugk, 2021-05-04:
Say e.g. I have a SeaBIOS or whatever… there likely still needs support by the vendor for the upgrade to work?

As with fwupd/LVFS there is a big documentation on all that stuff: https://fwupd.org/lvfs/docs/vendors

Here I see none of that. I mean if it is just a client… okay, but then link to the server component or so? Or spec?
After all, without having actual update files/upgrades from the vendors the whole project is kinda not… working… (I don’t dare to say useless), because well… you cannot upgrade anything without updates.

If something like this would be documented, one could go to some vendor and say “Hey, there is this cool project called _bootupd_ for updating your BIOS, don’t you want to support it too?” and in an ideal world, they’d say “Of course, we'll add that right away!”.
ANyway, one could and probably should try, especially when a vendor explicitly supports Linux.

travier, 2021-05-07:
I think there is a confusion here. This is for bootloader updates, such as GRUB, not BIOS or firmware updates.

The updates are newer versions of bootloaders (GRUB, etc.) as provided by the distribution.

rugk, 2021-05-07:
Ahhh… okay, then I was confused about the comparison to fwupd, as that is clearly about firmware updates or so.

closed 2021-05-11 by travier