coreos/bootupd #229: rework CI to generate synthetic RPMs

cgwalters, 2021-10-01:
See https://github.com/coreos/bootupd/pull/228 and https://github.com/coreos/bootupd/pull/203

Basically to test upgrades, we're downgrading to an older grub build which is broken.  We're just testing that the synthetic process of changing files actually works, but we're not currently testing rebooting into it.

Instead, we should download the working RPM and tweak it by:

 - adding files (safe)
 - Figuring out a way to mutate e.g. shim or grub binaries in a benign way that doesn't affect functionalty (this will (hopefully) not work with Secure Boot though), but at least we can reboot in non-SB UEFI
 -

bgilbert, 2021-10-01:
Once we have this, we can revert #228.

dralley, 2026-04-25:
So, the `rpm` crate that you already optionally depend on (which I maintain) has the ability to parse and build RPMs.

If you have a use case for rebuilding an RPM with modifications, I can see if I can accommodate that.  Relatedly I've heard some ideas from kernel folks that it would be nice to be able to build different variants of the kernel package with different sets of kernel modules, or perhaps rebuild in-place to strip out unneeded modules.  Kinda similar?