coreos/bootupd #367: `bootupctl backend install` leaks mount to EFI

jlebon, 2022-08-22:
Currently, running `bootupctl backend install` will fail to unmount the EFI partition after it completes. The `bootupd.service` unit has `MountFlags=slave`, but that doesn't apply to image building tools like cosa running the command directly.

jlebon, 2022-08-22:
Hmm for the cosa case, because we don't actually have systemd running in the supermin VM and we just do `reboot -f` at the end, I wonder if it's possible in theory to lose unsynced data.

Anyway, the workaround short-term for this is for cosa to run the command in a mount namespace. (Will add that along with other miscellaneous patches.)

cgwalters, 2022-09-30:
I think the `unshare -m` is actually the right fix for this in general; it means we're running the same way inside systemd as outside of it.

That said, https://github.com/coreos/bootupd/pull/379 also makes it cleaner for us to unmount the ESP when we're done too, and a patch to do so can be built after that lands.

closed 2022-10-04 by cgwalters