coreos/bootupd #378: Support for Fedora Silverblue?

xlejo, 2022-09-28:
Hi!

Currently, running `bootupd validate` or `bootupd update` on Fedora Siverblue 36 doesn't work. 

```
$ sudo bootupd update
error: internal error: Could not find "/dev/disk/by-partlabel/EFI-SYSTEM"
$ ls -l /dev/disk/by-partlabel/
total 0
lrwxrwxrwx. 1 root root 15 Sep 28 09:08 'EFI\x20System\x20Partition' -> ../../nvme0n1p1
```

There is a option to specified the EFI partition or is a expected behavior?

Thanks!

xlejo, 2022-09-29:
@davidstrauss You are right :(

```
$ ls -l /dev/disk/by-partlabel 
total 0
lrwxrwxrwx. 1 root root 15 Sep 29 14:47 EFI-SYSTEM -> ../../nvme0n1p1
$ sudo bootupctl status
No components installed.
No components are adoptable.
Boot method: EFI
$ sudo bootupctl adopt-and-update
No components are adoptable.
$ sudo bootupctl validate
No components installed.
$ sudo bootupctl update
No components installed.
```

xlejo, 2022-09-29:
If it helps, `fwupdmgr update` doesn't work either.

```
$ sudo fwupdmgr update
<truncated>
Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Decompressing…           [                                       ]Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/EFI/BOOT/BOOTX64.EFI Authenticode checksum [<skipped>] is present in dbx
$ rpm -qa | grep shim
shim-x64-15.6-2.x86_64
shim-ia32-15.6-2.x86_64
```

cgwalters, 2022-09-30:
Probably the best fix here is to look for a `/boot/efi` mount and use that if available.

xlejo, 2022-09-30:
@cgwalters The mounts seems to be ok.

```
$ mount | grep boot
/dev/nvme0n1p2 on /boot type ext4 (rw,relatime,seclabel)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=ascii,shortname=winnt,errors=remount-ro)
$ ls -l /boot/efi/EFI/*
/boot/efi/EFI/BOOT:
total 1536
-rwx------. 1 root root 1210776 Aug 18 09:54 BOOTX64.EFI
-rwx------. 1 root root  357248 Aug 18 09:54 fbx64.efi

/boot/efi/EFI/fedora:
total 7248
-rwx------. 1 root root     110 Aug 18 09:54 BOOTX64.CSV
drwx------. 2 root root    4096 Aug 18 09:54 fonts
drwx------. 2 root root    4096 Sep  8 09:34 fw
-rwx------. 1 root root   66952 Aug 18 09:54 fwupdx64.efi
-rwx------. 1 root root     144 Aug 18 09:54 grub.cfg
-rwx------. 1 root root 2536712 Aug 18 09:54 grubx64.efi
-rwx------. 1 root root 1159560 Aug 18 09:54 mmx64.efi
-rwx------. 1 root root 1210776 Aug 18 09:54 shim.efi
-rwx------. 1 root root 1210776 Aug 18 09:54 shimx64.efi
-rwx------. 1 root root 1204496 Aug 18 09:54 shimx64-fedora.efi
```

I'm doing something wrong?

davidstrauss, 2022-09-30:
@xlejo I suspect that @cgwalters is referring to how `bootupd` ought to find resources for adoption to achieve compatibility with Silverblue.

cgwalters, 2022-09-30:
Started on some prep cleanup for this in https://github.com/coreos/bootupd/pull/379

cgwalters, 2022-10-14:
PR in https://github.com/coreos/bootupd/pull/387

closed 2022-10-16 by cgwalters