coreos/bootupd #398: Add `bootupctl backend install` option to install BIOS bootloader

HuijingHei, 2022-11-11:
When build image, need to install bootloader for uefi firmware when running `bootupctl backend install` (see https://github.com/coreos/coreos-assembler/blob/main/src/create_disk.sh#L407), for BIOS/PReP bootloader need to call `grub2-install`. This is not consistent and convenient. 

What we want to do is add an option to cover BIOS/PReP firmware if run `bootupctl backend install`. See related issue https://github.com/coreos/coreos-assembler/issues/3156

cgwalters, 2022-11-14:
To implement this we're basically adding a new "component" in bootupd terms.  Until now, bootupd has multiple layers of abstraction with the idea that it might be managing multiple things ("components") - specifically on x86_64, it could be managing both "bios" and "uefi".

But the abstractions haven't really been tested because we've only had one.

I'd probably say to start with just x86_64+bios i.e. a new "bios" component.  

To handle grub2-install for ppc64le we'd need to expand the set of architectures this project builds for - and probably in theory PReP/ppc64le would be a new distinct component ("openfirmware")?

### Simpler alternative

Add a new "grub2" component that is basically a thin wrapper around `grub2-install` and just records the version number - there'd be just a dummy data payload.  The more I think about it the more this probably makes sense.

HuijingHei, 2022-11-15:
> Add a new "grub2" component that is basically a thin wrapper around `grub2-install` and just records the version number - there'd be just a dummy data payload. The more I think about it the more this probably makes sense.

This seems more simple and neat.

HuijingHei, 2022-11-16:
Just curious, if failed to boot after firmware updated, should we consider the rollback?

cgwalters, 2023-02-25:
This was done!

closed 2023-02-25 by cgwalters