coreos/bootupd #630: handle efi vendordir better

cgwalters, 2024-03-15:
See https://github.com/rhinstaller/anaconda/pull/5508#issuecomment-2000443423 and previous

Basically during an installation process we should really fix things so that we find the EFI vendor for the shim/grub we're going to install (from `/usr/lib/bootupd/updates/EFI`), verify its presence in the real ESP, and then target that.

jkonecny12, 2024-04-11:
I see this is closed. Does that mean that the bootupd installed by Anaconda will work on dual boot or more adjustments are required?

HuijingHei, 2024-04-15:
> Does that mean that the bootupd installed by Anaconda will work on dual boot or more adjustments are required?

Not quite sure, this issue is only to make sure to find the `vendor` directory (and shim path) and will not break creating the EFI boot manager entry. Maybe @cgwalters has more context about your question.

travier, 2024-04-18:
https://github.com/coreos/bootupd/pull/633 helped but is not complete. There are cases where could have multiple shim binaries in the EFI partition and we only want to touch our own, not others.

HuijingHei, 2024-04-19:
> #633 helped but is not complete. There are cases where could have multiple shim binaries in the EFI partition and we only want to touch our own, not others.
> 
> The logic should be:
> 
> * Look at what we have in the image
> * Find the corresponding in the EFI partition
> * Update only that

For example, firstly find the `<vendor>/shimx64.efi` file under `/usr/lib/bootupd/updates/EFI/`, then find it under EFI partition, if they are the same, then get the shim path; if not, will fail. Is this correct?

travier, 2024-04-19:
I've edited my comments to try to clarify it

HuijingHei, 2024-04-19:
Thanks @travier for the clarification!

closed 2024-05-15 by travier