coreos/bootupd #658: `bootupctl backend install --update-firmware` should match current Anaconda logic

travier, 2024-05-15:
See discussions in:
- https://bugzilla.redhat.com/show_bug.cgi?id=2268505
- https://github.com/rhinstaller/anaconda/pull/5508
- https://pagure.io/workstation-ostree-config/pull-request/453#comment-199664
- https://gitlab.com/fedora/ostree/sig/-/issues/1

HuijingHei, 2024-06-03:
What I understood is, according to https://github.com/rhinstaller/anaconda/pull/5508#issuecomment-1984219257

The current issue in bootupd:

- with `--update-firmware`, it runs `efibootmgr` with `--create` (which is the same as -c, it creates an entry and adds it to the boot order)
- without `--update-firmware`, it does not run efibootmgr at all.

There's no way to make it create an entry but not add it to the boot order.

**Change to:**

- keep with `--update-firmware` as it is
- without `--update-firmware`, run `efibootmgr -C` (`-C` creates an entry but does not add it to the boot order)

Or

- make `--update-firmware` as default to run `efibootmgr -c` to create an entry and add it to boot order
- if there is explicit `--update-firmware false`, run `efibootmgr -C` to create an entry but not add it to the boot order

travier, 2024-06-03:
We want to be able to use `bootupd backend install --update-firmware` in Anaconda to setup the boot order on EFI systems. This is https://github.com/rhinstaller/anaconda/pull/5508 and ideally we'll re-open and merge this PR once the issue is fixed here.

The issue is that when called with `--update-firmware`, bootupd currently removes the `BootCurrent` boot entry (https://github.com/coreos/bootupd/blob/19610be787cf2d027e3f3dccb556bba9c2e4b389/src/efi.rs#L459) and then adds a new `BootEntry` for the system being installed (https://github.com/coreos/bootupd/blob/19610be787cf2d027e3f3dccb556bba9c2e4b389/src/efi.rs#L497).

The current Anaconda behavior (and the one we want to reproduce in bootupd) is to only create a new boot entry and set it as the first one in the boot order (not deleting the `BootCurrent` boot entry).

Anaconda also creates the `BootEntry` with the name from the `EFI/fedora/BOOTX64.CSV` file in the ESP (Fedora for Fedora instead of `fedora` for bootupd).

So the first step is likely to delete https://github.com/coreos/bootupd/blob/19610be787cf2d027e3f3dccb556bba9c2e4b389/src/efi.rs#L459 and see what happens.

Then we'll have to add logic to read the name from `BOOTX64.CSV` to properly setup the boot entry.

That should be enough to let us enable that option in Anaconda again and then re-include bootupd in Atomic Desktops.

---

For a later issue, the idea is that we should consider implementing all the options (only those that make sense) from https://pykickstart.readthedocs.io/en/latest/kickstart-docs.html#bootloader so that the bootupd path in Anaconda using a kickstart config matches the one for the classic non-bootupd path.

travier, 2024-06-03:
Product name comes from: https://github.com/rhinstaller/anaconda/blob/6e4f976f25f9d7668fa62a7d3141735f9828ec6a/pyanaconda/core/product.py#L67

travier, 2024-06-03:
So several things:
- We need to decide where we read the boot entry name from. Either we keep reading it from the vendor dir (as bootupd does right now) or we need to read it from os-release for example. We can probably keep the existing logic for now.
- For the rework of the clear boot entry step, we should match boot entries without looking at the case to remove both `Fedora` and `fedora` for example.

travier, 2024-06-03:
The most important part for the removal is that we only remove "our" boot entries and not other boot entries. Better to remove less than too much.

HuijingHei, 2024-06-03:
> Boot0003* Fedora        HD(1,GPT,ae02f19e-99ce-4a8a-8d08-379c45b5048b,0x800,0x12c000)/\EFI\fedora\shimx64.efi

Thanks @travier for the clarification.
A little confused, the bootupd is using `fedora` (vendordir), why we see this as `Fedora`, does this mean it is using fb, which reads product name from `EFI/fedora/BOOTX64.CSV`?

HuijingHei, 2024-06-04:
> So several things:
> 
> * We need to decide where we read the boot entry name from. Either we keep reading it from the vendor dir (as bootupd does right now) or we need to read it from os-release for example. We can probably keep the existing logic for now.

Or maybe read from `BOOTX64.CSV` to align the name as `Fedora` with `anaconda` ?
> * For the rework of the clear boot entry step, we should match boot entries without looking at the case to remove both `Fedora` and `fedora` for example.

If we have the same boot entry name `Fedora`, then only clear the same boot entry name, does this make sense?

travier, 2024-06-04:
> A little confused, the bootupd is using `fedora` (vendordir), why we see this as `Fedora`

The output above is from a Fedora Kinoite 40 installation that was not done via bootupd but via Anaconda.

travier, 2024-06-04:
What's fb?

travier, 2024-06-04:
> Or maybe read from `BOOTX64.CSV` to align the name as `Fedora` with `anaconda` ?

We would have to get clarification on how the product names can be set for Anaconda as I don't think it comes from this place after more thoughts. We likely don't expect Fedora Remixes or variants to change the shim package to change branding.

> If we have the same boot entry name `Fedora`, then only clear the same boot entry name, does this make sense?

I think it's reasonable to remove both `Fedora` and `fedora` entries for the boot list. The case does not really matter here. This would let us keep our current setup and split this change into a later issue/PR.

HuijingHei, 2024-06-04:
> We would have to get clarification on how the product names can be set for Anaconda as I don't think it comes from this place after more thoughts. We likely don't expect Fedora Remixes or variants [...]

It is the `fallback.efi` that reads label from `BOOTX64.CSV`, see https://www.rodsbooks.com/efi-bootloaders/fallback.html, add comment https://bugzilla.redhat.com/show_bug.cgi?id=2268505#c20 to get pointer where `Fedora` comes.

> 
> I think it's reasonable to remove both `Fedora` and `fedora` entries for the boot list. The case does not really matter here. This would let us keep our current setup and split this change into a later issue/PR.

SGTM.

HuijingHei, 2024-06-04:
Anaconda gets the product name from `get_product_name()` which is from [product.py#L107](https://github.com/rhinstaller/anaconda/blob/6e4f976f25f9d7668fa62a7d3141735f9828ec6a/pyanaconda/core/product.py#L107), and value is from `ANACONDA_PRODUCTNAME` [product.py#L85](https://github.com/rhinstaller/anaconda/blob/6e4f976f25f9d7668fa62a7d3141735f9828ec6a/pyanaconda/core/product.py#L85), and definition `ANACONDA_PRODUCTNAME=$(sed -r -e 's/ *release.*//' /etc/system-release)` from [liveinst#L46](https://github.com/rhinstaller/anaconda/blob/f52880c8ac0f0b19d5e4b7887c80aedec1d4bfa7/data/liveinst/liveinst#L46), so it is from `/etc/system-release`, maybe can align it to bootupd in the later PR.

travier, 2024-06-10:
With https://github.com/coreos/bootupd/pull/665 merged, we can consider this one fixed.

travier, 2024-06-10:
Thanks @HuijingHei

closed 2024-06-10 by travier