coreos/bootupd #690: bootupd changed UEFI server boot order to left the bootc system boot first

henrywang, 2024-07-24:
When I run `bootc install to-existing-root` test on aarch64 bare metal server, the bootupd changed boot order to left the bootc system boot first. To re-install a new system, I have to re-configure the boot order to enable network boot first.
```
    Installing image: docker://quay.io/redhat_emp1/*****:u2d2
    Digest: sha256:1103ab6499bb1bf0f11632bb5afbf7f34e1dd02170127a8c3191de409635188a
    Initializing ostree layout
    Initializing sysroot
    ostree/deploy/default initialized as OSTree stateroot
    Deploying container image
    Loading usr/lib/ostree/prepare-root.conf
    Deployment complete
    Running bootupctl to install bootloader
    BootCurrent: 0005
    Timeout: 6 seconds
    BootOrder: 0005,0003,0004,0001,0000,0002,0006,0009
    Boot0000  fedora
    Boot0001* UEFI: Built-in EFI Shell
    Boot0002  redhat
    Boot0003* UEFI: PXE IPv4 Intel(R) Network D8:5E:D3:8F:A5:04
    Boot0004* UEFI: PXE IPv4 Intel(R) Network D8:5E:D3:8F:A5:05
    Boot0006  Fedora
    Boot0009  Gemini
    Boot0005* redhat
    Installed: grub.cfg
    Installed: "redhat/grub.cfg"
    Installation complete!
```

cgwalters, 2024-07-24:
The RFE here I think is for an option to preserve the existing boot order. Some use cases are likely relying on us replacing it.

henrywang, 2025-04-01:
Hi @cgwalters, I'm a little bit confused here. Why does the bootupd not replace the `Boot0000` which is created by package mode installation? But create a new duplicated one `Boot0002`? Thanks.

BootNext: 0000
BootCurrent: 0002
Timeout: 5 seconds
BootOrder: 0003,0004,0001,0000,0002
Boot0000* Red Hat Enterprise Linux	HD(1,GPT,aee15015-baad-4623-8d37-fd47ee4d4fac,0x800,0x12c000)/\EFI\REDHAT\SHIMAA64.EFI **(Created by beaker system installation)**
Boot0001* UEFI: Built-in EFI Shell	VenMedia(5023b95c-db26-429b-a648-bd47664c8012)0000424f
Boot0002* Red Hat Enterprise Linux
	HD(1,GPT,aee15015-baad-4623-8d37-fd47ee4d4fac,0x800,0x12c000)/\EFI\redhat\shimaa64.efi **(Created by bootupd)**
Boot0003* UEFI: PXE IPv4 Intel(R) Network D8:5E:D3:8F:A4:D0	PcieRoot(0x40000)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(d85ed38fa4d0,1)/IPv4(0.0.0.0,0,DHCP,0.0.0.0,0.0.0.0,0.0.0.0)0000424f
Boot0004* UEFI: PXE IPv4 Intel(R) Network D8:5E:D3:8F:A4:D1	PcieRoot(0x40000)/Pci(0x1,0x0)/Pci(0x0,0x1)/MAC(d85ed38fa4d1,1)/IPv4(0.0.0.0,0,DHCP,0.0.0.0,0.0.0.0,0.0.0.0)0000424f

HuijingHei, 2025-04-01:
Logically it should replace the same bootentry, will check it.