coreos/bootupd #707: Graceful recover after a failure

travier, 2024-08-22:
From https://github.com/coreos/bootupd/issues/694#issuecomment-2256146956:

If for whatever reason the previous command fails, then I need to `systemctl reset-failed` before calling bootupd again.

```
root@fedora:~# systemd-run --unit bootupd --property=MountFlags=slave --service-type=oneshot --pipe bootupctl status
Running as unit: bootupd.service
root@fedora:~# systemd-run --unit bootupd --property=MountFlags=slave --service-type=oneshot --pty bootupctl status
Running as unit: bootupd.service; invocation ID: c111d0e1cebe4401bb89af4cfa5358f1
Press ^] three times within 1s to disconnect TTY.

root@fedora:~# systemd-run --unit bootupd --property=MountFlags=slave --service-type=oneshot --pty bootupctl status
Failed to start transient service unit: Unit bootupd.service was already loaded or has a fragment file.
root@fedora:~# systemctl status bootupd.service
× bootupd.service - /usr/bin/bootupctl status
     Loaded: loaded (/run/systemd/transient/bootupd.service; transient)
  Transient: yes
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: signal) since Mon 2024-07-29 16:46:17 CEST; 14s ago
 Invocation: c111d0e1cebe4401bb89af4cfa5358f1
    Process: 4624 ExecStart=/usr/bin/bootupctl status (code=killed, signal=HUP)
   Main PID: 4624 (code=killed, signal=HUP)
   Mem peak: 3M
        CPU: 32ms

Jul 29 16:46:17 fedora systemd[1]: Starting bootupd.service - /usr/bin/bootupctl status...
Jul 29 16:46:17 fedora systemd[1]: bootupd.service: Main process exited, code=killed, status=1/HUP
Jul 29 16:46:17 fedora systemd[1]: bootupd.service: Failed with result 'signal'.
Jul 29 16:46:17 fedora systemd[1]: Failed to start bootupd.service - /usr/bin/bootupctl status.
root@fedora:~# systemd-run --unit bootupd --property=MountFlags=slave --service-type=oneshot --pty bootupctl status
Failed to start transient service unit: Unit bootupd.service was already loaded or has a fragment file.
```

travier, 2024-08-22:
A quick & dirty way to avoid that is issue is to always call `systemctl reset-failed bootupd.service` before trying to run another command.

HuijingHei, 2024-09-03:
Close as fixed by https://github.com/coreos/bootupd/pull/715

closed 2024-09-03 by HuijingHei