coreos/bootupd #717: Adoption failure on Atomic Desktops using BIOS

travier, 2024-09-05:
Trying to adopt / update an Atomic Desktop using BIOS boot fails:

```
$ bootupctl adopt-and-update -vvvv
[TRACE bootupd] executing cli
Running as unit: bootupd.service
[TRACE bootupd] executing cli
[TRACE bootupd::bootupd] No saved state
[TRACE bootupd::bootupd] Remaining known components: 2
[TRACE bootupd::bootupd] Not adoptable: BIOS
[DEBUG bootupd::efi] Unmounting
error: Failed to find ESP device
```

travier, 2024-09-05:
Looks like we never actually implement adoption for BIOS: [`870023e` (#420)](https://github.com/coreos/bootupd/pull/420/commits/870023eb6a7073d93194c8688b86444f3d74f775)

HuijingHei, 2024-09-18:
Look more about the error, find https://github.com/coreos/bootupd/pull/723 (included in `bootupd-202409180258.0.2.21.20.g4c5a514-1.fc41.x86_64`) can not fix the issue, boot silverblue rawhide using BIOS and run `bootupctl status` will also get the `error: Failed to find ESP device`.

I think the original designation is for hybrid partition layout which supports both BIOS and UEFI booting, so there is no such error for `fcos` or `rhcos`.

But for none hybrid partition disk, for example, when booting BIOS, there is no ESP device and so we hit the error.

```
fedora@fedora:~$ lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
zram0  251:0    0  3.8G  0 disk [SWAP]
vda    252:0    0   20G  0 disk 
├─vda1 252:1    0    1M  0 part 
├─vda2 252:2    0    1G  0 part /boot
└─vda3 252:3    0   19G  0 part /var/home
                                /var
                                /sysroot/ostree/deploy/fedora/var
                                /usr
                                /etc
                                /
                                /sysroot

fedora@fedora:~$ sudo bootupctl status -vvvvv
[TRACE bootupd] executing cli
Running as unit: bootupd.service
[TRACE bootupd] executing cli
[TRACE bootupd::bootupd] Gathering status for installed component: BIOS
[TRACE bootupd::bootupd] Remaining known components: 1
[DEBUG bootupd::efi] Unmounting
error: Failed to find ESP device

root@fedora:~# ls /dev/disk/by-label/
fedora_fedora
```

To fix this, on hybrid partition disk:
- booting with BIOS, skip gathering UEFI status if ESP device not exists
- booting with UEFI, skip gathering BIOS status (not sure about the condition)
Or just update the current boot firmware, but this will also prevent for the hybrid partition layout, maybe not suitable.

@travier any suggestions? Thanks!

travier, 2024-09-18:
Ideally, bootup would skip updating the EFI if the system is BIOS booted and it can not find an EFI partition.

cgwalters, 2024-09-18:
Yeah, when adopting I think we can gracefully handle the case where we don't find one or the other.

On CoreOS derivatives we could detect that it's a CoreOS and require finding both of them, but honestly at *this* point that's a real corner case because you'd have to have a really old system to need to go through that path.

travier, 2024-09-19:
Ouch, does this means we should not release 0.22.0 in Fedora as that will break EFI setups there?

closed 2024-09-24 by cgwalters

HuijingHei, 2024-09-25:
Got selinux avc denied logs using the new version:

```
----
time->Wed Sep 25 07:46:09 2024
type=AVC msg=audit(1727250369.837:244): avc:  denied  { search } for  pid=1920 comm="lsblk" name="udev" dev="tmpfs" ino=51 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:udev_var_run_t:s0 tclass=dir permissive=1
----
time->Wed Sep 25 07:46:09 2024
type=AVC msg=audit(1727250369.837:245): avc:  denied  { read } for  pid=1920 comm="lsblk" name="b252:0" dev="tmpfs" ino=1145 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:udev_var_run_t:s0 tclass=file permissive=1
----
time->Wed Sep 25 07:46:09 2024
type=AVC msg=audit(1727250369.837:246): avc:  denied  { open } for  pid=1920 comm="lsblk" path="/run/udev/data/b252:0" dev="tmpfs" ino=1145 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:udev_var_run_t:s0 tclass=file permissive=1
----
time->Wed Sep 25 07:46:09 2024
type=AVC msg=audit(1727250369.837:247): avc:  denied  { getattr } for  pid=1920 comm="lsblk" path="/run/udev/data/b252:0" dev="tmpfs" ino=1145 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:udev_var_run_t:s0 tclass=file permissive=1
```

HuijingHei, 2024-09-25:
The selinux issue is tracked by https://github.com/fedora-selinux/selinux-policy/issues/2362

cgwalters, 2024-09-26:
Yeah the attempt to confine bootupd is just causing us so many problems 😢