coreos/bootupd #739: Bootupd: thread 'main' panicked at src/bios.rs:116:62

TheGreatDeadOne, 2024-10-01:
**Describe the bug**
After updating bootupd, this bug appeared when trying to check the status of bootupd.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Update the system
2. Reboot
3. view bootupd status (or update)

**Screenshots**

[image]

**OS version:**
```
State: idle
BootedDeployment:
● fedora:fedora/rawhide/x86_64/silverblue
                  Version: Rawhide.20240930.n.0 (2024-09-30T06:07:39Z)
                   Commit: d44afe119fb862868666a1829781e2a1888497d4c0189186e290289cc35c369d
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944
```

**Additional context**
I ran it with RUST_BACKTRACE=full, but I didn't get any more information.

travier, 2024-10-01:
Can you try the workaround from https://github.com/coreos/bootupd/issues/738 to get a backtrace?

TheGreatDeadOne, 2024-10-01:
Here the backtrace:

```
Running as unit: bootupd.service
thread 'main' panicked at src/bios.rs:116:62:
JSON was not well-formatted: Error("invalid type: null, expected a string", line: 5, column: 23)
stack backtrace:
   0:     0x55f766b385a5 - <unknown>
   1:     0x55f766b602db - <unknown>
   2:     0x55f766b354af - <unknown>
   3:     0x55f766b39921 - <unknown>
   4:     0x55f766b395fc - <unknown>
   5:     0x55f766b39f81 - <unknown>
   6:     0x55f766b39de7 - <unknown>
   7:     0x55f766b38a69 - <unknown>
   8:     0x55f766b39a74 - <unknown>
   9:     0x55f7669699c3 - <unknown>
  10:     0x55f766969e36 - <unknown>
  11:     0x55f7669a9b61 - <unknown>
  12:     0x55f7669ae31e - <unknown>
  13:     0x55f7669aff25 - <unknown>
  14:     0x55f7669c6f6a - <unknown>
  15:     0x55f766978a66 - <unknown>
  16:     0x55f766978a5c - <unknown>
  17:     0x55f766b2de20 - <unknown>
  18:     0x55f7669dbd17 - <unknown>
  19:     0x7ff7326121c8 - __libc_start_call_main
  20:     0x7ff73261228b - __libc_start_main@GLIBC_2.2.5
  21:     0x55f76696a135 - <unknown>
  22:                0x0 - <unknown>

```

travier, 2024-10-02:
With debuginfo installed:
```
Running as unit: bootupd.service
thread 'main' panicked at src/bios.rs:116:62:
JSON was not well-formatted: Error("invalid type: null, expected a string", line: 5, column: 23)
stack backtrace:
   0:     0x558387fb95a5 - std::backtrace_rs::backtrace::libunwind::trace::h5c85e557799ed486
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/../../backtrace/src/backtrace/libunwind.rs:116:5
   1:     0x558387fb95a5 - std::backtrace_rs::backtrace::trace_unsynchronized::ha97b107185df65bb
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x558387fb95a5 - std::sys::backtrace::_print_fmt::h490acf9e9b8c6eb2
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/sys/backtrace.rs:65:5
   3:     0x558387fb95a5 - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::h9c32407e5a23c650
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/sys/backtrace.rs:40:26
   4:     0x558387fe12db - core::fmt::rt::Argument::fmt::hae324c745842212e
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/core/src/fmt/rt.rs:173:76
   5:     0x558387fe12db - core::fmt::write::h8e3a6cb8df1f9a95
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/core/src/fmt/mod.rs:1182:21
   6:     0x558387fb64af - std::io::Write::write_fmt::h83bcab37323a9399
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/io/mod.rs:1827:15
   7:     0x558387fba921 - std::sys::backtrace::BacktraceLock::print::hd3c35caa6032e632
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/sys/backtrace.rs:43:9
   8:     0x558387fba921 - std::panicking::default_hook::{{closure}}::hd3c6083514eb2656
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panicking.rs:269:22
   9:     0x558387fba5fc - std::panicking::default_hook::h94d20e9291e6eb42
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panicking.rs:296:9
  10:     0x558387fbaf81 - std::panicking::rust_panic_with_hook::hfa25182080856bef
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panicking.rs:800:13
  11:     0x558387fbade7 - std::panicking::begin_panic_handler::{{closure}}::h2fc3fd5367175cd3
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panicking.rs:674:13
  12:     0x558387fb9a69 - std::sys::backtrace::__rust_end_short_backtrace::h877093daaa72bd28
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/sys/backtrace.rs:168:18
  13:     0x558387fbaa74 - rust_begin_unwind
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panicking.rs:665:5
  14:     0x558387dea9c3 - core::panicking::panic_fmt::hfc4c464a0d356173
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/core/src/panicking.rs:74:14
  15:     0x558387deae36 - core::result::unwrap_failed::hed0c5a840d13d6d8
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/core/src/result.rs:1679:5
  16:     0x558387e2ab61 - core::result::Result<T,E>::expect::h4a278a4ffd3b9b1e
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/core/src/result.rs:1059:23
  17:     0x558387e2ab61 - bootupd::bios::Bios::get_bios_boot_partition::hf80eb676ef158f7d
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/bios.rs:116:32
  18:     0x558387e2ab61 - <bootupd::bios::Bios as bootupd::component::Component>::query_adopt::hbd7f47bf2d2f2ae6
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/bios.rs:168:44
  19:     0x558387e2f31e - bootupd::bootupd::status::h783212648de06519
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/bootupd.rs:311:34
  20:     0x558387e30f25 - bootupd::cli::bootupctl::CtlCommand::run_status::h47ffd5fa37e76902
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/cli/bootupctl.rs:104:17
  21:     0x558387e30f25 - bootupd::cli::bootupctl::CtlCommand::run::haabf88ed246e99ee
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/cli/bootupctl.rs:88:38
  22:     0x558387e47f6a - bootupd::cli::MultiCall::run::h3adeb0582e24cec0
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/cli/mod.rs:36:40
  23:     0x558387e47f6a - bootupd::run_cli::h3ec10f62211d0c8e
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/main.rs:68:11
  24:     0x558387e47f6a - bootupd::main::h830863b943d342cd
                               at /usr/src/debug/rust-bootupd-0.2.23-1.fc42.x86_64/src/main.rs:48:21
  25:     0x558387df9a66 - core::ops::function::FnOnce::call_once::h68b4f28c0539a3ac
[... cut ...]
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/panic.rs:350:14
  36:     0x558387faee20 - std::rt::lang_start_internal::he74de233149dbe8b
                               at /builddir/build/BUILD/rust-1.81.0-build/rustc-1.81.0-src/library/std/src/rt.rs:141:20
  37:     0x558387e5cd17 - main
  38:     0x7f680a6121c8 - __libc_start_call_main
  39:     0x7f680a61228b - __libc_start_main@GLIBC_2.2.5
  40:     0x558387deb135 - _start
  41:                0x0 - <unknown>
```

travier, 2024-10-02:
Reproduced on Silverblue Rawhide.20240930.n.0, UEFI only (does not happen with in a BIOS only VM).

travier, 2024-10-02:
Looks like pttype can be null but we dont account for that case: https://github.com/coreos/bootupd/blob/3e55890869eeea808f870bf82d8a5fbf5d1752df/src/bios.rs#L19
```
lsblk --json --output PATH,PTTYPE,PARTTYPENAME
{
   "blockdevices": [
      {
         "path": "/dev/sr0",
         "pttype": null,
         "parttypename": null
      },{
         "path": "/dev/zram0",
         "pttype": null,
         "parttypename": null
      },{
         "path": "/dev/vda",
         "pttype": "gpt",
         "parttypename": null
      },{
         "path": "/dev/vda1",
         "pttype": "gpt",
         "parttypename": "EFI System"
      },{
         "path": "/dev/vda2",
         "pttype": "gpt",
         "parttypename": "Linux extended boot"
      },{
         "path": "/dev/vda3",
         "pttype": "gpt",
         "parttypename": "Linux filesystem"
      },{
         "path": "/dev/mapper/luks-df2d5f95-5725-44dd-83e1-81bc4cdc49b8",
         "pttype": null,
         "parttypename": null
      }
   ]
}
```

closed 2024-10-04 by jmarrero