coreos/bootupd #748: lsblk: /dev/sdXX[/boot]: not a block device

gallium-stanza, 2024-10-18:
This is a fedora silverblue 41 system, installed by bootc-install-to-disk (not anaconda).
no fstab file, no installer-created nor user-created subvolumes, just the default bootc install.
all bootupctl command exit with the same error:
```
~$ sudo bootupctl status
Running as unit: bootupd.service
lsblk: /dev/sda3[/boot]: not a block device

error: Command {
    program: "lsblk",
    args: [
        "lsblk",
        "--paths",
        "--noheadings",
        "--output",
        "PKNAME",
        "/dev/sda3[/boot]",
    ],
    create_pidfd: false,
} failed with exit status: 32

```
```
~$ rpm -qa | grep bootupd
bootupd-0.2.24-1.fc41.x86_64
```

```
~$ cat /boot/bootupd-state.json
{"installed":{"EFI":{"meta":{"timestamp":"2024-08-08T12:14:11Z","version":"grub2-efi-ia32-1:2.12-4.fc41.x86_64,grub2-efi-x64-1:2.12-4.fc41.x86_64,shim-ia32-15.8-3.x86_64,shim-x64-15.8-3.x86_64"},"filetree":{"children":{"BOOT/BOOTIA32.EFI":{"size":747681,"sha512":"sha512:6d4396f289400516b883733f0fd3621e7ec4d70afd02e988651f37db81298775da69c04b998d87a4760b2fa4b96130c70eb0875fad1cf290c52ea606ae40d12e"},"BOOT/BOOTX64.EFI":{"size":949424,"sha512":"sha512:cc23d8c3cb2dcf749075268b77eb796fb430182cbbc04171ded14d43e32b4a5cdeeb1a08666ee0e288bd37d63f657a9af5e7f2012dd70694d11212d705c60b42"},"BOOT/fbia32.efi":{"size":70360,"sha512":"sha512:c3e94b8d6f07ec9c88b66d48e5b887d29288dd1c9914bf8226af7656e35a023c07a6dbf8a2875994485c348ca358591e02a8eb19574299db3ffb6486697a1568"},"BOOT/fbx64.efi":{"size":87816,"sha512":"sha512:d6107afdaebab6c6f3a9136334b54aeba089a8b40d53f38c74c791dca1274b9d6a02d744a634938516cedfa96a67cb0cff6a59c5537b36faf3c70e3bcee622af"},"fedora/BOOTIA32.CSV":{"size":112,"sha512":"sha512:53a8ec886d24c049ee27f2377de3f02951d537171be1ab8d11459453f094e60ff3667c2b7608581392283643b7864008b7cc4d1cdac6a3ebcab3d0d7d410d9e1"},"fedora/BOOTX64.CSV":{"size":110,"sha512":"sha512:0c29b8ae73171ef683ba690069c1bae711e130a084a81169af33a83dfbae4e07d909c2482dbe89a96ab26e171f17c53f1de8cb13d558bc1535412ff8accf253f"},"fedora/grubia32.efi":{"size":3022144,"sha512":"sha512:a81b24b44fdd646d036e963e54dcc76b9465798209f02c89b4cf528966ecd72dbcac78f733b3fd177f0f48091b7605a48eaff0663768ea72b962e123065a2b88"},"fedora/grubx64.efi":{"size":4066624,"sha512":"sha512:2981b7fe1c22969605104bab9ae6f45696c81412b0c08afb78827bc96dc8b53279661b4b9926679e864a7427bbf183169538aa78d3c29eafcc7468ffd70aa721"},"fedora/mmia32.efi":{"size":673992,"sha512":"sha512:cd2cd82de67d2176d212fbe99caa6ef621f971b0f49f1fe7430dd1d4f1bf822fe8f87fcbf5a23512dbb8a9c90c75d5e2ff0e8f910954c62749ef35237e48720a"},"fedora/mmx64.efi":{"size":848080,"sha512":"sha512:d62e003cae3321b3b0aea35c27868176a867b06a1a1eb3266efe84f72f01db6ce71c36073afc192500385743b5eb3d1f35519729b51d0326cebb1e3a04b24281"},"fedora/shim.efi":{"size":949424,"sha512":"sha512:cc23d8c3cb2dcf749075268b77eb796fb430182cbbc04171ded14d43e32b4a5cdeeb1a08666ee0e288bd37d63f657a9af5e7f2012dd70694d11212d705c60b42"},"fedora/shimia32.efi":{"size":747681,"sha512":"sha512:6d4396f289400516b883733f0fd3621e7ec4d70afd02e988651f37db81298775da69c04b998d87a4760b2fa4b96130c70eb0875fad1cf290c52ea606ae40d12e"},"fedora/shimx64.efi":{"size":949424,"sha512":"sha512:cc23d8c3cb2dcf749075268b77eb796fb430182cbbc04171ded14d43e32b4a5cdeeb1a08666ee0e288bd37d63f657a9af5e7f2012dd70694d11212d705c60b42"}}},"adopted-from":null}},"pending":null,"static-configs":{"timestamp":"2024-07-20T08:17:30Z","version":"bootupd-0.2.20-2.fc41.x86_64"}}
```

```
~$ lsblk -O /dev/sda
ALIGNMENT ID-LINK                      ID                       DISC-ALN DAX DISC-GRAN DISK-SEQ DISC-MAX DISC-ZERO FSAVAIL FSROOTS                                                                                              FSSIZE FSTYPE FSUSED FSUSE% FSVER GROUP HCTL       HOTPLUG KNAME LABEL      LOG-SEC MAJ:MIN MAJ MIN MIN-IO MODE       MODEL              MQ NAME   OPT-IO OWNER PARTFLAGS PARTLABEL  PARTN PARTTYPE                             PARTTYPENAME        PARTUUID                             PATH      PHY-SEC PKNAME PTTYPE PTUUID                                 RA RAND  REV RM RO ROTA RQ-SIZE SCHED SERIAL     SIZE   START STATE   SUBSYSTEMS     MOUNTPOINT MOUNTPOINTS                        TRAN   TYPE UUID                                 VENDOR   WSAME WWN                ZONED ZONE-SZ ZONE-WGRAN ZONE-APP ZONE-NR ZONE-OMAX ZONE-AMAX
        0 wwn-0x5000c5004e3d32fb       0x5000c5004e3d32fb              0   0        4K        1       0B         0                                                                                                                                                disk  0:0:0:0          0 sda                  512   8:0   8   0     4096 brw-rw---- ST500DM002-1BD142   1 sda         0 root                                                                                                                           /dev/sda     4096        gpt    05006d25-3560-4062-bb40-fda9219cd5c9 4096    1 KC45  0  0    1      64 bfq   Z3T46GHG 465.8G         running block:scsi:pci                                               sata   disk                                      ATA         0B 0x5000c5004e3d32fb none       0B         0B       0B       0         0         0
        0 wwn-0x5000c5004e3d32fb-part1 0x5000c5004e3d32fb-part1        0   0        4K        1       0B         0                                                                                                                                                disk                   0 sda1                 512   8:1   8   1     4096 brw-rw----                     1 ├─sda1      0 root            BIOS-BOOT      1 21686148-6449-6e6f-744e-656564454649 BIOS boot           0add3f62-00a1-428e-801d-f732b7cb5a32 /dev/sda1    4096 sda    gpt    05006d25-3560-4062-bb40-fda9219cd5c9 4096    1       0  0    1      64 bfq                1M    2048         block:scsi:pci                                                      part                                                  0B 0x5000c5004e3d32fb none       0B         0B       0B       0         0         0
        0 wwn-0x5000c5004e3d32fb-part2 0x5000c5004e3d32fb-part2        0   0        4K        1       0B         0                                                                                                                     vfat                 FAT32 disk                   0 sda2  EFI-SYSTEM     512   8:2   8   2     4096 brw-rw----                     1 ├─sda2      0 root            EFI-SYSTEM     2 c12a7328-f81f-11d2-ba4b-00a0c93ec93b EFI System          bd62c39c-6aa6-40fc-bdcc-7e31da3f28de /dev/sda2    4096 sda    gpt    05006d25-3560-4062-bb40-fda9219cd5c9 4096    1       0  0    1      64 bfq              512M    4096         block:scsi:pci                                                      part 2E51-C9AC                                        0B 0x5000c5004e3d32fb none       0B         0B       0B       0         0         0
        0 wwn-0x5000c5004e3d32fb-part3 0x5000c5004e3d32fb-part3        0   0        4K        1       0B         0    383G /ostree/deploy/default/var                                                                           465.3G btrfs   81.1G    17%       disk                   0 sda3  root           512   8:3   8   3     4096 brw-rw----                     1 └─sda3      0 root            root           3 4f68bce3-e8cd-4db1-96e7-fbcaf984b709 Linux root (x86-64) d154cd01-a86a-4723-8ec1-ea3351ccb81d /dev/sda3    4096 sda    gpt    05006d25-3560-4062-bb40-fda9219cd5c9 4096    1       0  0    1      64 bfq            465.3G 1052672         block:scsi:pci /sysroot   /var                                      part 0776fc3e-5911-4d3c-a0b0-ff568f73faaa             0B 0x5000c5004e3d32fb none       0B         0B       0B       0         0         0
                                                                                                                           /ostree/deploy/default/var                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           /sysroot/ostree/deploy/default/var                                                                                            [... cut ...]                                                                                                                                                                                                                                                                                                                                       /sysroot                                                                                                                                                                  
```

gallium-stanza, 2024-10-18:
Version 0.2.20-2.fc41.x86_64 (the one used at installation time) had no issues, not sure when this problem was introduced.

travier, 2024-10-21:
Thanks for the report.

> This is a fedora silverblue 41 system, installed by bootc-install-to-disk (not anaconda).

Note that this is unsupported and untested right now.

PR #792 (2024-12-05): bios.rs: pass --nofsroot to findmnt
This fixes #748
commit f959d656 (2024-12-05): bios.rs: pass --nofsroot to findmnt

This fixes #748
closed 2024-12-05 by cgwalters