coreos/bootupd #762: Adopt fails with error "removing ".btmp.fedora/mmia32.efi": No such file or directory (os error 2)"

jorgeml, 2024-11-04:
On a X86-84 system running Fedora Silverblue 41:

```
$ sudo bootupctl status
Running as unit: bootupd.service
No components installed.
Detected: EFI: unknown
Boot method: EFI

$ sudo bootupctl adopt-and-update
Running as unit: bootupd.service
error: Failed adopt and update: applying filesystem changes: removing ".btmp.fedora/mmia32.efi": No such file or directory (os error 2)

$ sudo bootupctl --version
bootupctl 0.2.24

```

HuijingHei, 2024-11-05:
Thanks for the reports.

Does this mean you have upgraded the OS from older version to Fedora Silverblue 41 latest?

Which version are you using for Fedora Silverblue 41? As I remembered, the latest version includes [bootloader-update.service](https://pagure.io/workstation-ostree-config/c/2181f67a2af8b1d34629151fe8b09ae834ebf7c4?branch=f41), which should automatically update bootloader during booting after upgrade, so you might do not need to update manually.

jorgeml, 2024-11-05:
Thanks @HuijingHei ! Yes, this was an upgraded system. I have been running Silverblue for a few releases.

I updated to the latest Silverblue compose (41.20241105.0 (2024-11-05T00:40:00Z)) and the error has changed:

```
$ sudo bootupctl adopt-and-update
Running as unit: bootupd.service
error: Failed adopt and update: applying filesystem changes: removing ".btmp.BOOT/fbia32.efi": No such file or directory (os error 2)

$ sudo bootupctl validate
Running as unit: bootupd.service
No components installed.
```

HuijingHei, 2024-11-06:
I can not reproduce when I upgrade to 41.20241105.0, maybe there is something different with your env, could you help to output your `lsblk` result?

```
fedora@fedora:~$ journalctl -b -u bootloader-update.service --no-page
Nov 06 01:32:38 fedora systemd[1]: Starting bootloader-update.service - Update bootloader on boot...
Nov 06 01:32:38 fedora bootupctl[920]: Adopted and updated: EFI: grub2-efi-ia32-1:2.12-10.fc41.x86_64,grub2-efi-x64-1:2.12-10.fc41.x86_64,shim-ia32-15.8-3.x86_64,shim-x64-15.8-3.x86_64
Nov 06 01:32:38 fedora systemd[1]: Finished bootloader-update.service - Update bootloader on boot.
fedora@fedora:~$ lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
zram0  251:0    0  3.8G  0 disk [SWAP]
vda    252:0    0   20G  0 disk 
├─vda1 252:1    0  600M  0 part /boot/efi
├─vda2 252:2    0    1G  0 part /boot
└─vda3 252:3    0 18.4G  0 part /var/home
                                /var
                                /sysroot/ostree/deploy/fedora/var
                                /usr
                                /etc
                                /
                                /sysroot
```

jorgeml, 2024-11-06:
@HuijingHei 

```
$ journalctl -b -u bootloader-update.service --no-page
Nov 06 17:33:47 deneb systemd[1]: Starting bootloader-update.service - Update bootloader on boot...
Nov 06 17:33:47 deneb bootupctl[1222]: error: Failed adopt and update: applying filesystem changes: removing ".btmp.BOOT/BOOTIA32.EFI": No such file or directory (os error 2)
Nov 06 17:33:47 deneb systemd[1]: bootloader-update.service: Main process exited, code=exited, status=1/FAILURE
Nov 06 17:33:47 deneb systemd[1]: bootloader-update.service: Failed with result 'exit-code'.
Nov 06 17:33:47 deneb systemd[1]: Failed to start bootloader-update.service - Update bootloader on boot.
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 465.8G  0 disk 
├─sda1        8:1    0   100M  0 part 
├─sda2        8:2    0    16M  0 part 
├─sda3        8:3    0 232.1G  0 part 
└─sda4        8:4    0   715M  0 part 
sdb           8:16   0 931.5G  0 disk 
sdc           8:32   0 931.5G  0 disk 
zram0       252:0    0     8G  0 disk [SWAP]
nvme0n1     259:0    0 931.5G  0 disk 
├─nvme0n1p1 259:1    0   600M  0 part /boot/efi
├─nvme0n1p2 259:2    0     1G  0 part /boot
└─nvme0n1p3 259:3    0 929.9G  0 part /var/home
                                      /var
                                      /sysroot/ostree/deploy/fedora/var
                                      /usr
                                      /etc
                                      /
                                      /sysroot
```

@travier 

```
$ sudo ls -alhR /boot/efi
/boot/efi:
total 12K
drwx------. 3 root root 4.0K Jan  1  1970 .
drwxr-xr-x. 7 root root 4.0K Nov  6 17:33 ..
drwx------. 5 root root 4.0K Nov  6 17:33 EFI

/boot/efi/EFI:
total 20K
drwx------. 5 root root 4.0K Nov  6 17:33 .
drwx------. 3 root root 4.0K Jan  1  1970 ..
drwx------. 2 root root 4.0K Sep  9  2022 BOOT
drwx------. 2 root root 4.0K Sep  9  2022 .btmp.BOOT
drwx------. 2 root root 4.0K Nov  6 17:36 fedora

/boot/efi/EFI/BOOT:
total 1.0M
drwx------. 2 root root 4.0K Sep  9  2022 .
drwx------. 5 root root 4.0K Nov  6 17:33 ..
-rwx------. 1 root root 928K Apr 10  2024 BOOTX64.EFI
-rwx------. 1 root root  86K Apr 10  2024 fbx64.efi

/boot/efi/EFI/.btmp.BOOT:
total 1.0M
drwx------. 2 root root 4.0K Sep  9  2022 .
drwx------. 5 root root 4.0K Nov  6 17:33 ..
-rwx------. 1 root root 928K Apr 10  2024 BOOTX64.EFI
-rwx------. 1 root root  86K Apr 10  2024 fbx64.efi

/boot/efi/EFI/fedora:
total 6.5M
drwx------. 2 root root 4.0K Nov  6 17:36 .
drwx------. 5 root root 4.0K Nov  6 17:33 ..
-rwx------. 1 root root  110 Apr 10  2024 BOOTX64.CSV
-rwx------. 1 root root 8.1K Nov  6 17:33 grub.cfg
-rwx------. 1 root root 8.1K Nov  5 17:55 grub.cfg.old
-rwx------. 1 root root 1.0K Nov  6 17:36 grubenv
-rwx------. 1 root root 3.8M Apr 10  2024 grubx64.efi
-rwx------. 1 root root 829K Apr 10  2024 mmx64.efi
-rwx------. 1 root root 928K Apr 10  2024 shim.efi
-rwx------. 1 root root 928K Apr 10  2024 shimx64.efi

```

jorgeml, 2024-11-06:
Sorry, forgot this one:

```
$ sudo bootupctl adopt-and-update -vvvv
[TRACE bootupd] executing cli
Running as unit: bootupd.service
[TRACE bootupd] executing cli
[TRACE bootupd::bootupd] No saved state
[TRACE bootupd::bootupd] Remaining known components: 2
[DEBUG bootupd::bios] Skip BIOS adopt
[TRACE bootupd::bootupd] Not adoptable: BIOS
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] No EFI variable LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::efi] No EFI variable StubInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::component] No CoreOS aleph detected
[DEBUG bootupd::efi] Unmounting
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] No EFI variable LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::efi] No EFI variable StubInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::component] No CoreOS aleph detected
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] applying adoption diff: additions: 0 removals: 6 changes: 1
[DEBUG bootupd::filetree] Copy fedora to .btmp.fedora
[DEBUG bootupd::efi] Unmounting
error: Failed adopt and update: applying filesystem changes: removing ".btmp.fedora/shimia32.efi": No such file or directory (os error 2)
```

HuijingHei, 2024-11-07:
> /boot/efi/EFI/BOOT:
> total 1.0M
> drwx------. 2 root root 4.0K Sep  9  2022 .
> drwx------. 5 root root 4.0K Nov  6 17:33 ..
> -rwx------. 1 root root 928K Apr 10  2024 BOOTX64.EFI
> -rwx------. 1 root root  86K Apr 10  2024 fbx64.efi
> 
> /boot/efi/EFI/fedora:
> total 6.5M
> drwx------. 2 root root 4.0K Nov  6 17:36 .
> drwx------. 5 root root 4.0K Nov  6 17:33 ..
> -rwx------. 1 root root  110 Apr 10  2024 BOOTX64.CSV
> -rwx------. 1 root root 8.1K Nov  6 17:33 grub.cfg
> -rwx------. 1 root root 8.1K Nov  5 17:55 grub.cfg.old
> -rwx------. 1 root root 1.0K Nov  6 17:36 grubenv
> -rwx------. 1 root root 3.8M Apr 10  2024 grubx64.efi
> -rwx------. 1 root root 829K Apr 10  2024 mmx64.efi
> -rwx------. 1 root root 928K Apr 10  2024 shim.efi
> -rwx------. 1 root root 928K Apr 10  2024 shimx64.efi

Thanks for the testing, this is helpful. Seems the system does not have `shim-ia32` and `grub2-efi-ia32`, but this is expected as we do not need it on `x64`, if remove the related file, the update would fail

```
root@fedora:/boot/efi/EFI# ls BOOT/
BOOTIA32.EFI  BOOTX64.EFI   fbia32.efi    fbx64.efi     
root@fedora:/boot/efi/EFI# rm -f BOOT/fbia32.efi BOOT/BOOTIA32.EFI
root@fedora:~# bootupctl adopt-and-update
Running as unit: bootupd.service
error: Failed adopt and update: applying filesystem changes: removing ".btmp.BOOT/fbia32.efi": No such file or directory (os error 2)
```

We need to ignore the removing if can not find the file, also curious how we could remove `shim-ia32` and `grub2-efi-ia32` packages, maybe remove directly the related files under `/boot/efi` ?

jorgeml, 2024-11-07:
Thanks @HuijingHei ! I think I might have caused this when fixing a boot error a year or two ago, one of those which had an article in Fedora Magazine. I will wait for the new release.

travier, 2024-11-07:
Some (arguably broken) EFI systems need 32bits bootloaders binaries to boot an x86_64 one, thus why those are installed by default now in Fedora but may not be on existing systems. In most cases they are indeed not needed but I don't know if there is a way to know for sure.

travier, 2024-11-07:
So it looks like bootupd should not fail when removing a file if it already does not exist.

travier, 2024-11-07:
https://github.com/coreos/bootupd/blob/main/src/filetree.rs#L350 looks like this one either should not fail if the file already does not exists or we should test if the file exists before calling it.

Verhoeckx, 2024-11-07:
Hello Everybody,

I have the same problem!

I will give as much information as possible.

- Have been running Fedora Silverblue for a couple of years.
- Last update was from Fedora Silverblue 40 to 41.
<br>

sudo bootupctl adopt-and-update -vvvv
```
[TRACE bootupd] executing cli
Running as unit: bootupd.service
[TRACE bootupd] executing cli
[TRACE bootupd::bootupd] No saved state
[TRACE bootupd::bootupd] Remaining known components: 2
[DEBUG bootupd::bios] Skip BIOS adopt
[TRACE bootupd::bootupd] Not adoptable: BIOS
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] No EFI variable LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::efi] No EFI variable StubInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::component] No CoreOS aleph detected
[DEBUG bootupd::efi] Unmounting
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] No EFI variable LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::efi] No EFI variable StubInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
[TRACE bootupd::component] No CoreOS aleph detected
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[DEBUG bootupd::efi] Reusing existing "/boot/efi"
[TRACE bootupd::efi] applying adoption diff: additions: 0 removals: 6 changes: 6
[DEBUG bootupd::filetree] Copy fedora to .btmp.fedora
[DEBUG bootupd::efi] Unmounting
error: Failed adopt and update: applying filesystem changes: removing ".btmp.fedora/BOOTIA32.CSV": No such file or directory (os error 2)

```

<br>

sudo systemctl reset-failed bootupd.service
```
Failed to reset failed state of unit bootupd.service: 
Unit bootupd.service not loaded.
```

<br>
sudo bootupctl status

```
Running as unit: bootupd.service
No components installed.
Detected: EFI: unknown
Boot method: EFI
```
<br>
sudo bootupctl validate

```
Running as unit: bootupd.service
No components installed.
```
<br>

Information about my system:
```
System:
  Kernel: 6.11.5-300.fc41.x86_64 arch: x86_64 bits: 64 compiler: gcc
    v: 2.43.1-2.fc41
  Desktop: GNOME v: N/A wm: gnome-shell dm: N/A Distro: Fedora Linux 41
    (Toolbx Container Image)
Machine:
  Type: Laptop System: Dell product: XPS 13 9380 v: N/A
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: Dell model: 0KTW76 v: A00 serial: <superuser required> part-nu: 08AF
    UEFI: Dell v: 1.26.0 date: 09/11/2023
Battery:
  ID-1: BAT0 charge: 41.0 Wh (96.7%) condition: 42.4/52.0 Wh (81.6%)
    volts: 8.5 min: 7.6 model: SMP DELL G8VCF6C serial: <filter>
    status: charging
  Device-1: hidpp_battery_1 model: Logitech Signature M650 L
    serial: <filter> charge: N/A status: discharging
CPU:
  Info: quad core model: Intel Core i7-8565U bits: 64 type: MT MCP
    arch: Comet/Whiskey Lake note: check rev: C cache: L1: 256 KiB L2: 1024 KiB
    L3: 8 MiB
  Speed (MHz): avg: 700 min/max: 400/4600 cores: 1: 700 2: 700 3: 700 4: 700
    5: 700 6: 700 7: 700 8: 700 bogomips: 31999
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel WhiskeyLake-U GT2 [UHD Graphics 620] vendor: Dell
    driver: i915 v: kernel arch: Gen-9.5 ports: active: DP-1,eDP-1
    empty: DP-2,DP-3 bus-ID: 00:02.0 chip-ID: 8086:3ea0
  Device-2: Microdia Integrated_Webcam_HD driver: uvcvideo type: USB
    rev: 2.0 speed: 480 Mb/s lanes: 1 bus-ID: 1-5:2 chip-ID: 0c45:6723
  Display: wayland server: N/A compositor: gnome-shell driver: X:
    loaded: modesetting unloaded: fbdev,vesa dri: iris gpu: i915 display-ID: 0
  Monitor-1: DP-1 model: Dell U2718Q res: 3840x2160 dpi: 160
    diag: 702mm (27.6")
  Monitor-2: eDP-1 model: AU Optronics 0x282b res: 3840x2160 dpi: 333
    diag: 336mm (13.2")
  API: EGL Message: EGL data requires eglinfo. Check --recommends.
Audio:
  Device-1: Intel Cannon Point-LP High Definition Audio vendor: Dell
    driver: snd_hda_intel v: kernel bus-ID: 00:1f.3 chip-ID: 8086:9dc8
  API: ALSA v: k6.11.5-300.fc41.x86_64 status: kernel-api
Network:
  Device-1: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter
    vendor: Rivet Networks Killer 1435 Wireless-AC driver: ath10k_pci v: kernel
    pcie: speed: 2.5 GT/s lanes: 1 bus-ID: 02:00.0 chip-ID: 168c:003e
    temp: 39.0 C
  IF: wlp2s0 state: up mac: <filter>
Bluetooth:
  Device-1: Foxconn / Hon Hai driver: btusb v: 0.8 type: USB rev: 1.1
    speed: 12 Mb/s lanes: 1 bus-ID: 1-7:3 chip-ID: 0489:e0a2
  Report: rfkill ID: hci0 rfk-id: 0 state: up address: see --recommends
Drives:
  Local Storage: total: 476.94 GiB used: 402.3 GiB (84.4%)
  ID-1: /dev/nvme0n1 vendor: Samsung model: PM981 NVMe 512GB
    size: 476.94 GiB speed: 31.6 Gb/s lanes: 4 serial: <filter> temp: 37.9 C
Partition:
  Message: No partition data found.
Swap:
  ID-1: swap-1 type: zram size: 8 GiB used: 0 KiB (0.0%) priority: 100
    dev: /dev/zram0
Sensors:
  System Temperatures: cpu: 50.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Info:
  Memory: total: 16 GiB note: est. available: 15.27 GiB used: 3.47 GiB (22.7%)
  Processes: 342 Power: uptime: 16m wakeups: 0 Init: systemd v: 256
    default: graphical
  Packages: pm: rpm pkgs: N/A note: see --rpm Compilers: N/A Shell: Bash
    v: 5.2.32 running-in: conmon inxi: 3.3.36
```

<br>
lsblk

```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
zram0       252:0    0     8G  0 disk [SWAP]
nvme0n1     259:0    0 476,9G  0 disk 
├─nvme0n1p1 259:1    0   600M  0 part /boot/efi
├─nvme0n1p2 259:2    0     1G  0 part /boot
└─nvme0n1p3 259:3    0 475,4G  0 part /var/home
                                      /var
                                      /sysroot/ostree/deploy/fedora/var
                                      /usr
                                      /etc
                                      /
                                      /sysroot

```

closed 2024-11-15 by HuijingHei

HuijingHei, 2024-11-21:
The new release [rust-bootupd-0.2.25-1.fc41](https://koji.fedoraproject.org/koji/buildinfo?buildID=2587263) would fix this issue.

Verhoeckx, 2024-11-27:
When can I expect this new package to be available?

Verhoeckx, 2024-11-28:
Mmm, it seem that the new package doesn't solve my problem.

My original problem was this:
[Can't update 'Secure Boot dbx Configuration Update](https://discussion.fedoraproject.org/t/cant-update-secure-boot-dbx-configuration-update/135566)  and @travier pointed out that my problem was related to the problem mentioned here.

This is what I just did:
```
rpm-ostree override replace  https://kojipkgs.fedoraproject.org//packages/rust-bootupd/0.2.25/1.fc41/x86_64/bootupd-0.2.25-1.fc41.x86_64.rpm
reboot
fwupdmgr update
# And got the following message:
Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/EFI/fedora/shimx64-fedora.efi Authenticode checksum [0ce02100f67c7ef85f4eed368f02bf7092380a3c23ca91fd7f19430d94b00c19] is present in dbx
```
As you can see, I still get an error message.
<br>

When I run:
```
sudo bootupctl adopt-and-update -vvvv
```
I get:
```
[TRACE bootupd] executing cli
Running as unit: bootupd.service
[TRACE bootupd] executing cli
[TRACE bootupd::bootupd] Gathering status for installed component: EFI
[DEBUG bootupd::efi] Unmounting
[TRACE bootupd::bootupd] Remaining known components: 1
[DEBUG bootupd::bios] Skip BIOS adopt
[TRACE bootupd::bootupd] Not adoptable: BIOS
No components are adoptable.
```
My firmware is (seen as) BIOS and not UEFI? Do I need to wait for BIOS support?

Verhoeckx, 2024-11-30:
Okay, will do that!