coreos/bootupd #766: support dynamically extending/adding update payloads

cgwalters, 2024-11-07:
See this https://github.com/ondrejbudai/fedora-bootc-raspi/blob/cbff66a8bdb82b6276814765ffdb82c4ec4efa9c/bootupctl-shim#L7

Basically today the bootloader update payload is often part of the base image, but we need to support dynamically extending it with layered image content as well.

---

In the short term, what may suffice is to offer an explicit user-hookable phase that can copy things from the container to `/boot` as part of `bootc install to-filesystem`.

HuijingHei, 2024-11-20:
Is this something that teach bootupctl to read [config](https://containers.github.io/bootc/man-md/bootc-install-config.html) for example: `/usr/lib/bootc/install/00-firmware.toml`
```
[install.firmware]
type = "bootc-raspi-firmwares"
```
then it will get `install.fimware.type`, and find `/usr/lib/bootc-raspi-firmwares/`, finally copy the files to default directory `/boot/efi`.

Or maybe make bootc to read the config file and get `install.fimware.type`, then pass the env to `bootupctl` ?

Not sure if my understanding is correct. Any suggestions? Thanks!

cgwalters, 2024-11-20:
So one thing heavily related to this is today rpm-ostree in "unified core" mode automatically redirects content in `/boot` to `/usr/lib/ostree-boot` - so that means the files here:

```
rpm -ql bcm2711-firmware
/boot/efi/bcm2711-rpi-4-b.dtb
/boot/efi/bcm2711-rpi-400.dtb
/boot/efi/bcm2711-rpi-cm4-io.dtb
/boot/efi/bcm2711-rpi-cm4.dtb
/boot/efi/bcm2711-rpi-cm4s.dtb
/boot/efi/bcm2712-rpi-5-b.dtb
/boot/efi/bcm2712-rpi-cm5-cm4io.dtb
/boot/efi/bcm2712-rpi-cm5-cm5io.dtb
/boot/efi/bcm2712-rpi-cm5l-cm4io.dtb
/boot/efi/bcm2712-rpi-cm5l-cm5io.dtb
/boot/efi/bcm2712d0-rpi-5-b.dtb
/boot/efi/fixup4.dat
/boot/efi/fixup4cd.dat
/boot/efi/fixup4db.dat
/boot/efi/fixup4x.dat
/boot/efi/start4.elf
/boot/efi/start4cd.elf
/boot/efi/start4db.elf
/boot/efi/start4x.elf
```

But `dnf` does no such thing, so...one simple way to handle this might actually just be changing `bootupctl backend install` to be idempotent - in the case where the base image already has an `/usr/lib/bootupd/updates/EFI.json` we extend (regenerating) it.

In theory though, it would be nicer to represent e.g. shim+grub as a distinct payload from these other blobs?

Doing that though...hmm, I guess the cleanest way would be for shim+grub to explicitly flag themselves as a unit, or we teach bootc to recognize those directly?

alexlarsson, 2025-03-05:
For automotive we need more than just put stuff in /boot. For example, if android boot is used, then the boot dir will contain an aboot.img file, and this needs to be written to the boot_a partition. We do ship tools to do the writing, but we need to have a place to hook in callouts to those commands with the right arguments.

cgwalters, 2025-03-05:
Yeah, bootupd abstracts today over the (GRUB) MBR/BIOS case which is a dedicated partition.

But this is all hardcoded (in Rust) in this project. Adding more bootloaders here I think wouldn't be hard, and the code was definitely designed to enable it.

But, maybe we could in theory try to rework things such that we have `/usr/lib/bootupd/backends` - in the *simplest* case of a single bootloader state then bootupd would end up being a really thin wrapper.

(Unfortunately we do need to think about cases like disk images bootable with both grub BIOS as well as UEFI, so there are conceptually two. But OTOH today that case is grub for both, so maybe we could collapse that into just "grub" conceptually and always update both)

kfox1111, 2025-04-25:
Has there been any progress on this? interested in simplifying bootc based images on rpi's. Need some way of copying the rpi boot bits onto the boot partiion when its mounted.

cgwalters, 2025-05-28:
To be clear in this there's two very different cases, though ideally we make them one:

- Content shipped in an RPM with files in `/boot` that are *present in the initial base image build* where `bootupctl backend generate-update-payload` is invoked
- Content added after

So the most ideal thing here would be to again define a generic/extensible/pluggable payload mechanism, and then the cases fold together.

cgwalters, 2025-06-05:
One might wonder: "Why is this any harder than `dnf install bcm2711-firmware`"? 

The answer is basically that the rationale for bootupd is that we don't have a mechanism to *transactionally* update the bootloaders (again see the README.md) and in general it makes sense to split management of the bootloader state from management of the operating system.
They are different things.

`dnf` does no such thing of course today but *it doesn't have transactional updates and rollbacks* so it doesn't matter.

Related to this, bootupd puts all the content of bootloader state in `/usr`, and only copies it into `/boot/efi` when `bootupctl update` is invoked. That's why bootupd has this whole concept of "payloads".

Today, bootc does not automatically run bootupd, but of course some operating systems are shipping the example `bootloader-update.service` from this repo.

kfox1111, 2025-06-09:
@cgwalters generally sounds like a good plan to me.

One use case I'm trying to explore is using bootc with a rpi5's with m.2 drives only.

So far the only firmware I know of that works with it is:
https://github.com/worproject/rpi5-uefi

So having some mechanism for dropping that at the root of the boot partition (not /efi) would be very useful, so it can chainload grub.

Would that work with this proposed mechanism, or would that need to be handled some other way?

cgwalters, 2025-06-09:
@kfox1111 without digging fairly deep (only had time to glance at the website) I think the answer is no, the most current subthread here is just about extending things on the ESP.

Though is what rpi5 needs different than what original person who filed this issue (see the top) needed? That's also just extending the ESP I think.

But for sure we should support ~arbitrary changes to call this overall issue complete which basically has to cover what you're asking for right? (I am not too familiar with the details of rpi booting but this overall

kfox1111, 2025-06-09:
Yeah, the other issue was closed as a dup of this. but if this one isn't going to handle non /boot/EFI, then the other one is not a dup. I'm ok either way, just trying to make sure we don't drop the use case.

The pi just requires some files to exist on the root of the first partition /dev/nvme0n1p1 to load an EFI firmware so it can honor whats in /boot/EFI. The pi's don't come with a native EFI environment unfortunately.

cgwalters, 2025-06-09:
Yeah, this issue should cover that one too when completed.

HuijingHei, 2025-06-10:
> The `extend-boot-efi` verb would pull content out of `/boot/efi` and automatically translate it to the (proposed in fedora) new layout of `/usr/lib/efi/<name>/<version>/EFI/<contents...>`, and then from _that_ we need to translate it into bootupd's current concept of the EFI payload

SGTM.

say-paul, 2025-06-10:
> The extend-boot-efi verb would pull content out of /boot/efi and automatically translate it to the (proposed in fedora) new layout of /usr/lib/efi/<name>/<version>/EFI/<contents...>

I am skeptical if the layout makes sense when someone wants to put things in /boot
 The workflow I propose:
1. We can create a method `bootupctl extend-boot-payload   <payload>`  payload: /boot, /boot/efi, and copy things to /usr/lib/bootupd/update 
2. extend `generate-update-metadata()` to generate metadata:  `/usr/lib/bootupd/updates/boot.json`
3. extend `bootupd backend install`  to move the files from `/usr/lib/bootupd/update` to `/boot` , update the bootupd-state.json with the filetree

say-paul, 2025-06-11:
> Does it make sense to put the efi files to /boot/efi/BOOT

Not at all, I  suggested `/usr/lib/bootupd/update/BOOT/*` to `/boot` not  `/boot/efi/`.

Since grub+shim transactions are done in /boot/efi/EFI/<centos/fedora/rhel>
We need a extend payload mechanism for to extend for /boot/{files/sub-directories}.

say-paul, 2025-06-11:
We can also split the payload extension mechanism to:
- boot_payload_extension : /boot/{files,sub-directories} except /boot/efi , meta BOOT.JSON path: /usr/lib/bootupd/update/BOOT/
- esp_payload_extension:  only /boot/efi : meta ESP.json: path  /usr/lib/bootupd/update/ESP/ , though it can be merged with EFI.json and   /usr/lib/bootupd/update/EFI/ - but I don't know how will it react with existing system.

say-paul, 2025-06-23:
> The extend-boot-efi verb would pull content out of /boot/efi and automatically translate it to the (proposed in fedora) new layout of /usr/lib/efi/<name>/<version>/EFI/<contents...> , and then from that we need to translate it into bootupd's current concept of the EFI payload 

I did this in the #935 , with bit of modification and generalization of accepting input from user, so 
1. we can run `bootupd extend-payload-to-esp /boot/efi` internally
2. user can do  in containerfile `RUN bootupctl backend extend-payload-to-esp <any_path>`

supakeen, 2026-09-17:
Note that the previously listed workaround is now also broken. I think the switch from `bubblewrap` to `ChrootCommand` here: https://github.com/bootc-dev/bootc/commit/c0ff1c9f made it so that the mounts are now non-recursive so the `bootupd` wrapper doesn't actually see `/boot/efi` and writes directly into `/boot/efi` (the not-mountpoint but the subdirectory on XBOOTLDR).

This means we have no way anymore to work around this issue for things that need additional firmware in their ESP.

cgwalters, 2026-09-17:
Ouch sorry about that @supakeen 

This one needs some careful design, and it just hasn't come to the top yet. The good news is this project is a bit revitalized and we're making momentum on other issues (although, the fix for one of those is what caused this of course)

ericcurtin, 2026-09-20:
Not the same but somewhat related, was asked to link https://github.com/bootc-dev/bootc/pull/2475