coreos/bootupd #786: Update does not add missing files in `/boot/efi/`

HuijingHei, 2024-12-02:
The context is https://github.com/coreos/bootupd/pull/765#issuecomment-2476213020, when doing updates, find that it does not add the missing files.

Steps:

- Fresh install silverblue 41.1.4 (older version), remove `BOOT/BOOTIA32.EFI & BOOT/fbia32.efi` and upgrade bootupd to `0.2.25-1.fc41`, remove `/boot/bootupd-state.json`, run `bootupctl adopt-and-update` successfully.
- Upgrade to 41.20241201.0, check `bootloader-update.service` is successful to update grub2 and shim, but check `BOOT/BOOTIA32.EFI & BOOT/fbia32.efi` are not added.

travier, 2024-12-02:
Maybe we should think of this as "Adoption does not record missing files in the ESP".

HuijingHei, 2024-12-03:
> Maybe we should think of this as "Adoption does not record missing files in the ESP".

When do adoption, seems we add all the version we have in `/usr/lib/bootupd/updates/EFI.json` to `/boot/bootupd-state.json`, also includes the package that are not installed, for example: `grub2-efi-ia32-1:2.12-15.fc41.x86_64,shim-ia32-15.8-3.x86_64`. If the adoption only include the installed packages in `bootupd-state.json`, then for the **next upgrade** it would add the missing files.
But want to confirm that if it is necessary to do the change for adoption to only include the installed packages `bootupd-state.json`?

travier, 2024-12-03:
> But want to confirm that if it is necessary to do the change for adoption to only include the installed packages `bootupd-state.json`?

Yes, I think that's what we should do. The adoption phase should only include in `bootupd-state.json` the files that actually exists in the ESP.

travier, 2024-12-03:
And maybe directly install the missing ones.

HuijingHei, 2024-12-03:
> And maybe directly install the missing ones.

Agree that it would make adopt simple, and finally sync with `update`, only concern if it will make customers / users confused that we add additional files that they might not need.