coreos/bootupd #810: grub2 vs grub name

SkyWar-design, 2025-01-01:
Hello, thank you for your project. I attempted to create a bootc-compatible image not based on Fedora and encountered several issues. The most significant challenge was the hardcoded references to grub2 in the code. The problem is that not all Linux-based systems adhere to the grub/grub2 naming convention. For instance, in the system I was working with, there is no grub2 package—what exists is simply grub, which is already version 2.*.

As a result, I had to manually replace all references to grub2 with grub in the code.

Would it be possible to make the dependency on grub2 optional? For example, this could be configured through a settings file or provided as a runtime option.

Thank you for considering this suggestion.

cgwalters, 2025-01-02:
Hi, indeed there are a lot of distribution-specific things that need to be abstracted in this project.

As far as not depending on grub, that topic strongly relates to https://github.com/ostreedev/ostree/pull/3359 which would lead us closer to supporting the type1 BLS, which would be the abstract interface - and would make it clearer how to use systemd-boot.

However, what specific bootloader are you planning to use?

SkyWar-design, 2025-01-02:
> Hi, indeed there are a lot of distribution-specific things that need to be abstracted in this project.
> 
> As far as not depending on grub, that topic strongly relates to [ostreedev/ostree#3359](https://github.com/ostreedev/ostree/pull/3359) which would lead us closer to supporting the type1 BLS, which would be the abstract interface - and would make it clearer how to use systemd-boot.
> 
> However, what specific bootloader are you planning to use?

Hello, thank you for your response. I will be using grub, which is actually the grub2 version. To be precise, here is the repository:

https://packages.altlinux.org/en/sisyphus/srpms/grub/

I know it looks strange, but it might not be the only Linux build that makes such renaming by abandoning the old version of grub. I tried creating symlinks from grub2 to grub, but it caused an error that I have already forgotten.

By replacing the code in the files

```
    src/grubconfigs.rs (grub2 → grub)
    src/efi.rs (["loader", "grub2"] → ["loader", "grub"])
    src/bios.rs (grub2-install → grub-install)
```

and manually building bootupd, I successfully installed bootc.

However, because I built bootupd manually, I had to apply a hack in the query_files function of the file src/packagesystem.rs:
  
```
    !rpmout.status.success()
        return Ok(ContentMetadata { timestamp: chrono::Utc::now(), version: "unknown".to_string() });
```

If the first steps are understandable, the last change in the packagesystem file looks the worst, and I don't know what it will lead to; I am still testing the resulting outcome.