coreos/bootupd #868: F42: SELinux denials when running bootloader-update.service

miabbott, 2025-03-07:
On Fedora 42 IoT, I observed the following denials:

```
Mar 06 15:15:17 localhost audit[959]: AVC avc:  denied  { execute } for  pid=959 comm="bootupctl" name="ostree" dev="overlay" ino=3113 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:install_exec_t:s0 tclass=file permissive=1
Mar 06 15:15:17 localhost audit[959]: AVC avc:  denied  { read open } for  pid=959 comm="bootupctl" path="/usr/bin/ostree" dev="overlay" ino=3113 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:install_exec_t:s0 tclass=file permissive=1
Mar 06 15:15:17 localhost audit[959]: AVC avc:  denied  { execute_no_trans } for  pid=959 comm="bootupctl" path="/usr/bin/ostree" dev="overlay" ino=3113 scontext=system_u:system_r:bootupd_t:s0 tcontext=system_u:object_r:install_exec_t:s0 tclass=file permissive=1
```

This doesn't seem to have affected the operation of the service, but it would be good to squash these.

Versions:

```
$ rpm -q bootupd selinux-policy
bootupd-0.2.26-3.fc42.x86_64
selinux-policy-41.33-1.fc42.noarch
```

travier, 2025-03-07:
The overall consensus on our side is that confining bootupd is not worth it: https://github.com/fedora-selinux/selinux-policy/pull/2257#issuecomment-2317052496

It is thus currently running as permissive on Fedora CoreOS & Atomic Desktops. SELinux issues should be reported to the SELinux policy repo upstream.

miabbott, 2025-03-10:
Closing per comment

closed 2025-03-10 by miabbott