coreos/bootupd #877: Failed to configure on s390x machine

henrywang, 2025-03-10:
Run `podman run --rm --privileged --pid=host -v /:/target -v /dev:/dev -v /var/lib/containers:/var/lib/containers --security-opt label=type:unconfined_t localhost/bootc:beaker bootc install to-existing-root --target-transport containers-storage` on s390x machine.

Failed with error:
```
ERROR Installing to filesystem: Listing partitions of /dev/dasda: Subprocess failed: ExitStatus(unix_wait_status(256))
sfdisk: /dev/dasda: does not contain a recognized partition table
```

HuijingHei, 2025-03-10:
Have a short talk with @henrywang , seems on s390x bootc / bootupd use `sfdisk` to [list partitions](https://github.com/bootc-dev/bootc/blob/main/lib/src/install.rs#L1849) that introduces the problem, might use `fdasd` according to [coreos-installer](https://github.com/coreos/coreos-installer/tree/main/src/s390x) 

```
[root@ibm-z-511 ~]# sfdisk -J /dev/dasda
sfdisk: /dev/dasda: does not contain a recognized partition table
```

cgwalters, 2025-03-24:
I think you're right - we'd need to copy the dasd code over to bootc probably and then vendor it here in bootupd?

Wait actually this error is just from bootc. bootupd is a no-op on s390x right now.

cgwalters, 2025-03-24:
We should probably transfer this issue to bootc?

HuijingHei, 2025-03-24:
> Wait actually this error is just from bootc. bootupd is a no-op on s390x right now.

You are right, agree to move this to bootc.

cgwalters, 2025-03-24:
Done in https://github.com/bootc-dev/bootc/issues/1226