coreos/bootupd #890: bootc-image-builder should not require PLATFORM_ID in /etc/os-release

yjftsjthsd-g, 2025-03-27:
(This is somewhat related to https://github.com/coreos/bootupd/issues/468 but is separate.)

As of today (`build_revision: 7dd9bea`), bootc-image-builder requires that the image it's building from has `PLATFORM_ID` in `/etc/os-release`; if this is absent, then:
```
Message: Generating manifest manifest-qcow2.json
2025/03/27 22:47:43 error: cannot build manifest: missing PLATFORM_ID in os-release
```

This is a problem for any aspirations the tool has of working with other distros (since `PLATFORM_ID` is apparently a Fedora invention), but even Fedora has a proposal to drop it - https://fedoraproject.org/wiki/Changes/Drop_PLATFORM_ID - which would obviously break things.

cgwalters, 2025-03-28:
Hi, I completely agree but this issue is unrelated to this project, can you please file this against https://github.com/osbuild/bootc-image-builder/issues ?

closed 2025-03-28 by cgwalters

yjftsjthsd-g, 2025-03-28:
Oh, it's under osbuild, that's why I didn't find it. Sorry, thought that was part of this project; will do!