coreos/bootupd #891: Two more s390x errors

henrywang, 2025-03-28:
Just found another two different errors on s390x machine:

1. `ERROR Installing to filesystem: Installing bootloader: Failed to find PReP partition with GUID 9E1A2D38-C612-4316-AA26-8B49521E5A8B` on `ibm-ic922-12.lab.eng.brq2.redhat.com`
2. `ERROR Installing to filesystem: Found multiple parent devices /dev/sda and /dev/nvme0n1; not currently supported` on `rdma-cert-03-lp10.rdma.lab.eng.rdu2.redhat.com`

cgwalters, 2025-03-28:
> ERROR Installing to filesystem: Installing bootloader: Failed to find PReP partition with GUID 9E1A2D38-C612-4316-AA26-8B49521E5A8B on ibm-ic922-12.lab.eng.brq2.redhat.com

Are you sure that wasn't on a ppc64le machine?

henrywang, 2025-06-25:
Yes, it's for ppc64le.