coreos/bootupd #896: bootupctl set-password

champtar, 2025-04-04:
When bootupd is used, Anaconda/kickstart `bootloader --password` is ignored

As pointed by Jiri Konecny: https://issues.redhat.com/browse/RHEL-78299?focusedId=26599031&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-26599031
it would be better to have a bootupctl sub command to handle managing the bootloader password

I was thinking of `bootupctl set-password` that takes either clear text or already hashed grub password and write user.cfg with `GRUB2_PASSWORD=...`

Any opinions ?

cgwalters, 2025-04-04:
I think the better fix here is to make `grub2-setpassword` work right?

champtar, 2025-04-04:
`grub2-set-password` does work now, that's why I've put this line here
https://github.com/coreos/bootupd/blob/aa5a2e8502bf3310dea2421a96633e75b5ee3501/src/grub2/configs.d/01_users.cfg#L1-L2

@cgwalters if you think that's out of scope for bootupd can you go comment in https://issues.redhat.com/browse/RHEL-78299 ?

champtar, 2025-04-07:
@cgwalters do you think it make sense to have a `bootupctl set-password` subcommand ? If not we just need to tell Anaconda team not to wait for it

champtar, 2025-04-08:
https://issues.redhat.com/browse/RHEL-78299?focusedId=26948559&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-26948559

From @cgwalters:
> bootupd was not intended as a complete abstraction over bootloaders, as I don't think such a thing would be possible; it'd be too leaky.
> It is just an abstraction over upgrading their code.
>
> So there's no password functionality in bootupd; any more than there's functionality for changing the UX timeout etc.

closed 2025-04-08 by champtar