coreos/bootupd #902: bootupctl: Failed to start transient service unit: Connection reset by peer

dustymabe, 2025-04-09:
When upgrading a 4.18 (RHEL 9.4) system to 4.19 (RHEL 9.6) after the update bootupctl status can't run:

```
[core@cosa-devsh ~]$ rpm -q bootupd
bootupd-0.2.27-3.el9.x86_64
[core@cosa-devsh ~]$ sudo bootupctl status 
Failed to start transient service unit: Connection reset by peer
```

The transition here is:

```
bootupd 0.2.18-1.el9 -> 0.2.27-3.el9
```

The test I did was boot `rhcos-418.94.202504080525-0-qemu.x86_64.qcow2` and then upgrade to `419.96.202504090041-0`. After upgrade run status command and observe failure.

cgwalters, 2025-04-09:
I'm not immediately reproducing this on a rhel-bootc system upgraded from rhel-bootc:9.4 to rhel-bootc:9.6. Can you get a bit more info about this from the journal? Also try `env SYSTEMD_LOG_LEVEL=debug bootupctl status`

HuijingHei, 2025-04-10:
I start 418.94.202504072147-0 and upgrade to 419.96.202504090041-0, can not reproduce. 

```
[root@cosa-devsh core]# rpm-ostree rebase ostree-unverified-registry:<419.96.202504090041-0>
  bootupd 0.2.18-1.el9 -> 0.2.27-3.el9
```
After reboot

```
[root@cosa-devsh core]# bootupctl status
Running as unit: bootupd.service
Component BIOS
  Installed: grub2-tools-1:2.06-86.el9_4.2.x86_64
  Update: Available: grub2-tools-1:2.06-104.el9_6.x86_64
Component EFI
  Installed: grub2-efi-x64-1:2.06-86.el9_4.2.x86_64,shim-x64-15.8-4.el9_3.x86_64
  Update: Available: grub2-efi-x64-1:2.06-104.el9_6.x86_64,shim-x64-15.8-3.el9_4.x86_64
No components are adoptable.
CoreOS aleph version: 418.94.202504072147-0
Boot method: BIOS

[root@cosa-devsh core]# rpm -q bootupd
bootupd-0.2.27-3.el9.x86_64
[root@cosa-devsh core]# bootupctl update -vvvv
...
[root@cosa-devsh core]# bootupctl status
Running as unit: bootupd.service
Component BIOS
  Installed: grub2-tools-1:2.06-104.el9_6.x86_64
  Update: At latest version
Component EFI
  Installed: grub2-efi-x64-1:2.06-104.el9_6.x86_64,shim-x64-15.8-3.el9_4.x86_64
  Update: At latest version
No components are adoptable.
CoreOS aleph version: 418.94.202504072147-0
Boot method: BIOS

[root@cosa-devsh core]# reboot
[root@cosa-devsh core]# bootupctl status
Running as unit: bootupd.service
Component BIOS
  Installed: grub2-tools-1:2.06-104.el9_6.x86_64
  Update: At latest version
Component EFI
  Installed: grub2-efi-x64-1:2.06-104.el9_6.x86_64,shim-x64-15.8-3.el9_4.x86_64
  Update: At latest version
No components are adoptable.
CoreOS aleph version: 418.94.202504072147-0
Boot method: BIOS
```

HuijingHei, 2025-04-10:
And can not reproduce when using `rhcos-418.94.202504080525-0` to upgrade to `419.96.202504090041-0`, not sure if there is any special config?

dustymabe, 2025-04-10:
OK. I was able to reproduce again:

```
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.16 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-a
Bus n/a: changing state RUNNING → CLOSING
Failed to start transient service unit: Connection reset by peer
Bus n/a: changing state CLOSING → CLOSED
```

Here are my reproduction steps:

```
cosa init ...
cosa buildfetch --url https://releases-rhcos--prod-pipeline.apps.int.prod-stable-spoke1-dc-iad2.itup.redhat.com/storage/prod/streams/4.18-9.4/builds --build=418.94.202504080525-0 --artifact=qemu
cosa decompress
cosa buildfetch --url https://releases-rhcos--prod-pipeline.apps.int.prod-stable-spoke1-dc-iad2.itup.redhat.com/storage/prod/streams/4.19-9.6/builds --build=419.96.202504090041-0 --artifact=ostree

cosa run -c --build=418.94.202504080525-0

sudo bootupctl status

oci=/var/mnt/workdir/builds/419.96.202504090041-0/x86_64/rhcos-419.96.202504090041-0-ostree.x86_64.ociarchive
sudo rpm-ostree rebase --experimental --reboot "ostree-unverified-image:oci-archive:${oci}"

# after reboot

sudo env SYSTEMD_LOG_LEVEL=debug bootupctl status
```

which shows for me:

```
Red Hat Enterprise Linux CoreOS 419.96.202504090041-0 4.19
SSH host key: SHA256:GZy75QDtmG2MNGb9JS/ufam93nxpH8NU7jaLOSirKeQ (RSA)
SSH host key: SHA256:ztwaDx7nluSWj5/d5OxU2dsOoABX2nW8uEUGb26Gflo (ECDSA)
SSH host key: SHA256:t63Q2bU4z2K+3Lr1V+9lc2vhHp0izOCSJqHBp3sL+K8 (ED25519)
ens4: 10.0.2.15 fe80::44ac:4cdf:26d3:38b
Ignition: ran on 2025/04/10 14:26:05 UTC (at least 2 boots ago)
Ignition: user-provided config was applied
cosa-devsh login: core (automatic login)

Last login: Thu Apr 10 14:27:29 on tty1
Red Hat Enterprise Linux CoreOS 419.96.202504090041-0
  Part of OpenShift 4.19, RHCOS is a Kubernetes-native operating system
  managed by the Machine Config Operator (`clusteroperator/machine-config`).

WARNING: Direct SSH access to machines is not recommended; instead,
make configuration changes via `machineconfig` objects:
  https://docs.openshift.com/container-platform/4.19/architecture/architecture-rhcos.html

---
[core@cosa-devsh ~]$
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.16 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-a
Bus n/a: changing state RUNNING → CLOSING
Failed to start transient service unit: Connection reset by peer
Bus n/a: changing state CLOSING → CLOSED
```

The journal doesn't really show anything of importance.

One thing I did find as part of reproducing this again is that if I run `cosa run` without the `-c` i.e. not the serial console, but SSH, it seems to work OK.

Some issue with the session of the serial console?

dustymabe, 2025-04-10:
Here's what it looks like when I execute the same steps without the `-c` on the `cosa run` command:

<details>

```
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.23 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=609 reply_cookie=2 signature=o error-name=n/a error-message=n/a
Running as unit: bootupd.service
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=AddMatch cookie=3 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=4 reply_cookie=0 signature=s error-name=a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=617 reply_cookie=4 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=org.freedesktop.DBus destination=:1.23 path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=NameAcquired cookie=4294967295 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.23 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=3 signature= error-name=n/a error-message=n/a
Match type='signal',sender='org.freedesktop.systemd1',path='/org/freedesktop/systemd1/unit/bootupd_2eservice',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged' successfully installed.
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=618 reply_cookie=0 signature=sa{sv}as error-name=n/a era
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=5 reply_cookie=0 signature=s error-name=a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=627 reply_cookie=5 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=619 reply_cookie=0 signature=sa{sv}as error-name=n/a era
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=6 reply_cookie=0 signature=s error-name=a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=629 reply_cookie=6 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=620 reply_cookie=0 signature=sa{sv}as error-name=n/a era
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=7 reply_cookie=0 signature=s error-name=a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=631 reply_cookie=7 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=621 reply_cookie=0 signature=sa{sv}as error-name=n/a era
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=8 reply_cookie=0 signature=s error-name=a
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=633 reply_cookie=8 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=624 reply_cookie=0 signature=sa{sv}as error-name=n/a era
[... cut ...] gnature=sa{sv}as error-name=n/a era
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/bootupd_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=11 reply_cookie=0 signature=s error-namea
Got message type=method_return sender=:1.0 destination=:1.23 path=n/a interface=n/a member=n/a cookie=641 reply_cookie=11 signature=a{sv} error-name=n/a error-message=n/a
Bus n/a: changing state RUNNING → CLOSED
```

</details.

dustymabe, 2025-04-10:
ok more info..

this has nothing to do with the upgrade (which is good, because it makes the bug less complicated).

I just thought it had to do with the upgrade because when I tested it without the upgrade the other day I ran `cosa run` without the `-c` (i.e. I wasn't interested in the bootup logs flowing to the console because I wasn't doing an upgrade).

The new reproducer is just boot 4.19 with `cosa run -c`. 

```
cosa run -c --build=419.96.202504090041-0
sudo env SYSTEMD_LOG_LEVEL=debug bootupctl status
```

At this point maybe this isn't something we need to investigate in the context of bootupd. I thought at first it was some weird interaction with the upgrade.

cgwalters, 2025-04-10:
Happy to do this myself if need be but I think the next step is looking for selinux denials, my best bet here is that `systemd-run` breaks when forwarding a physical (`/dev/tty*` with label `tty_device_t`) as opposed to virtual console (`/dev/pts` with label `user_devpts_t).

dustymabe, 2025-04-10:
@cgwalters bingo. I swear I should have thought of that, but there was no denial in the journal. I guess maybe there is a rule in there that is noaudit or something. 

If I `sudo setenforce 0` I can then run `sudo bootupctl status`.

I guess I need to file this as a bug against RHEL 9.6.

Thank you for pointing me in the right direction.

dustymabe, 2025-04-10:
hmm. though, maybe I spoke too soon.

If I manually `systemd-run` it works OK. Here is an example running `echo "success"`:

```
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug systemd-run -t echo "success"
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.42 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-a
Got message type=method_return sender=:1.0 destination=:1.42 path=n/a interface=n/a member=n/a cookie=854 reply_cookie=2 signature=o error-name=n/a error-message=n/a
Running as unit: run-u42.service
Press ^] three times within 1s to disconnect TTY.
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=AddMatch cookie=3 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du42_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=4 reply_cookie=0 signature=s error-nama
Got message type=method_return sender=:1.0 destination=:1.42 path=n/a interface=n/a member=n/a cookie=868 reply_cookie=4 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=org.freedesktop.DBus destination=:1.42 path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=NameAcquired cookie=4294967295 reply_cookie=0 signature=s error-name=n/a error-message=n/a
success
Got message type=method_return sender=org.freedesktop.DBus destination=:1.42 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=3 signature= error-name=n/a error-message=n/a
Match type='signal',sender='org.freedesktop.systemd1',path='/org/freedesktop/systemd1/unit/run_2du42_2eservice',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged' successfully installed.
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/run_2du42_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=859 reply_cookie=0 signature=sa{sv}as error-name=n/a a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du42_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=5 reply_cookie=0 signature=s error-nama
Got message type=method_return sender=:1.0 destination=:1.42 path=n/a interface=n/a member=n/a cookie=872 reply_cookie=5 signature=a{sv} error-name=n/a error-message=n/a
Bus n/a: changing state RUNNING → CLOSED
```

and here is a run of `bootupctl status` (note that since it's running in systemd already bootupctl won't try to re-exec itself with `systemd-run`:

```
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug systemd-run -t bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.44 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-a
Got message type=method_return sender=:1.0 destination=:1.44 path=n/a interface=n/a member=n/a cookie=884 reply_cookie=2 signature=o error-name=n/a error-message=n/a
Running as unit: run-u44.service
Press ^] three times within 1s to disconnect TTY.
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=AddMatch cookie=3 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du44_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=4 reply_cookie=0 signature=s error-nama
Got message type=method_return sender=:1.0 destination=:1.44 path=n/a interface=n/a member=n/a cookie=898 reply_cookie=4 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=org.freedesktop.DBus destination=:1.44 path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=NameAcquired cookie=4294967295 reply_cookie=0 signature=s error-name=n/a er [... cut ...]
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-name=n/a error-message=n/a
Bus n/a: changing state RUNNING → CLOSING
Failed to start transient service unit: Connection reset by peer
Bus n/a: changing state CLOSING → CLOSED
```

Maybe it's some issue with the [systemd unit settings](https://github.com/coreos/bootupd/blob/604866ea4f5743c2db5cb4706fa076d6e4c3d2ba/src/cli/bootupctl.rs#L13-L20) bootupctl is setting?

dustymabe, 2025-04-10:
ahh yep:

```
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug systemd-run --pipe bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a errora
Got message type=method_return sender=org.freedesktop.DBus destination=:1.54 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 ra
Bus n/a: changing state RUNNING → CLOSING
Failed to start transient service unit: Connection reset by peer
Bus n/a: changing state CLOSING → CLOSED
```

but then setting SELinux to permissive:

```
[core@cosa-devsh ~]$ sudo setenforce 0
[core@cosa-devsh ~]$ 
[core@cosa-devsh ~]$ sudo env SYSTEMD_LOG_LEVEL=debug systemd-run --pipe bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a errora
Got message type=method_return sender=org.freedesktop.DBus destination=:1.57 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=2 ra
Got message type=method_return sender=:1.0 destination=:1.57 path=n/a interface=n/a member=n/a cookie=994 reply_cookie=2 signature=o error-name=n/a error-message=n/a
Running as unit: run-u57.service
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=AddMatch cookie=3 reply_cookie=0 signature=s erroa
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du57_2eservice interface=org.freedesktop.DBus.Properties member=GetAla
Got message type=method_return sender=:1.0 destination=:1.57 path=n/a interface=n/a member=n/a cookie=1010 reply_cookie=4 signature=a{sv} error-name=n/a error-message=n/a
Component BIOS
  Installed: grub2-tools-1:2.06-104.el9_6.x86_64
  Update: At latest version
Component EFI
  Installed: grub2-efi-x64-1:2.06-104.el9_6.x86_64,shim-x64-15.8-3.el9_4.x86_64
  Update: At latest version
No components are adoptable.
CoreOS aleph version: 419.96.202504090041-0
Boot method: BIOS
Got message type=signal sender=org.freedesktop.DBus destination=:1.57 path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=NameAcquired cookie=4294967295 reply_cookie=0 signatua
Got message type=method_return sender=org.freedesktop.DBus destination=:1.57 path=n/a interface=n/a member=n/a cookie=4294967295 reply_cookie=3 signature= error-name=n/a error-message=n/a
Match type='signal',sender='org.freedesktop.systemd1',path='/org/freedesktop/systemd1/unit/run_2du57_2eservice',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged' succe.
Got message type=signal sender=:1.0 destination=n/a path=/org/freedesktop/systemd1/unit/run_2du57_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=1001 ra
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du57_2eservice interface=org.freedesktop.DBus.Properties member=GetAla
Got message type=method_return sender=:1.0 destination=:1.57 path=n/a interface=n/a member=n/a cookie=1014 reply_cookie=5 signature=a{sv} error-name=n/a error-message=n/a
Bus n/a: changing state RUNNING → CLOSED
```

dustymabe, 2025-04-10:
So the open question is:

1. change bootupd OR
2. chase a fix in SELinux

cgwalters, 2025-04-10:
I think both actually make sense

HuijingHei, 2025-04-11:
Nice finding!

Seems the problem is selinux issue but can not find any denied logs in `ausearch`.
```
[root@cosa-devsh core]# systemd-run --pipe echo "hello"
Failed to start transient service unit: Connection reset by peer

[root@cosa-devsh core]# setenforce 0
[root@cosa-devsh core]# systemd-run --pipe echo "hello"
Running as unit: run-u65.service
hello
```

dustymabe, 2025-04-11:
> I think both actually make sense

Opened a bug for the policy side: https://issues.redhat.com/browse/RHEL-86925

HuijingHei, 2025-04-16:
A little confused that can not reproduce this on FCOS 41.20250409.20.1

```
$ cosa run --qemu-image data/fedora-coreos-41.20250409.20.1-qemu.x86_64.qcow2 -m 4096 -c

[root@cosa-devsh core]# env SYSTEMD_LOG_LEVEL=debug systemd-run --pipe bootupctl status
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.29 path=n/a interface=n/a member=n/a  cookie=4294967295 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.DBus.Properties member=Get cookie=2 reply_cookie=0 signature=ss error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=762 reply_cookie=2 signature=v error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartTransientUnit cookie=3 reply_cookie=0 signature=ssa(sv)a(sa(sv)) error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=772 reply_cookie=3 signature=o error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=Get cookie=4 reply_cookie=0 signature=ss error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=780 reply_cookie=4 signature=v error-name=n/a error-message=n/a
Running as unit: run-u29.service
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=AddMatch cookie=5 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=6 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=790 reply_cookie=6 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=org.freedesktop.DBus destination=:1.29 path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=NameAcquired  cookie=4294967295 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.29 path=n/a interface=n/a member=n/a  cookie=4294967295 reply_cookie=5 signature= error-name=n/a error-message=n/a
Match type='signal',sender='org.freedesktop.systemd1',path='/org/freedesktop/systemd1/unit/run_2du29_2eservice',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged' successfully installed.
Got message type=signal sender=:1.1 destination=n/a path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged  cookie=781 reply_cookie=0 signature=sa{sv}as error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=7 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=792 reply_cookie=7 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.1 destination=n/a path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged  cookie=782 reply_cookie=0 signature=sa{sv}as error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=8 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=794 reply_cookie=8 signature=a{sv} error-name=n/a error-message=n/a
Got message type=signal sender=:1.1 destination=n/a path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged  cookie=783 reply_cookie=0 signature=sa{sv}as error-name=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=9 reply_cookie=0 signature=s error-name=n/a error-message=n/a
[... cut ...] e=n/a error-message=n/a
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1/unit/run_2du29_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=14 reply_cookie=0 signature=s error-name=n/a error-message=n/a
Got message type=method_return sender=:1.1 destination=:1.29 path=n/a interface=n/a member=n/a  cookie=810 reply_cookie=14 signature=a{sv} error-name=n/a error-message=n/a
Bus n/a: changing state RUNNING → CLOSED
```

HuijingHei, 2025-04-27:
When running `cosa run -c` with serial console, run `bootupctl status` failed and get error:
```
[core@cosa-devsh ~]$ sudo bootupctl status
Failed to start transient service unit: Connection reset by peer
```

This is because of selinux blocking `systemd-run --pipe` from serial console, see https://issues.redhat.com/browse/RHEL-86925, and the workaround is disable selinux using `sudo setenforce 0`, or add `-t/--pty` like `sudo systemd-run -t bootupctl status`.

We will wait for the selinux fix, and leave this as a known corner case issue that few people will hit.

closed 2025-04-27 by HuijingHei