coreos/bootupd #926: Extend `generate-update-metadata` to read from `/usr` (Unification of boot loader updates, phase 1)

travier, 2025-05-15:
Extend the `generate-update-metadata` logic to not query the RPM db but instead pick the versions from the new paths where the grub2 & shim binaries will be installed in `/usr/lib/(shim|grub)/%{version}-%{release}/EFI/...` when we land https://fedoraproject.org/wiki/Changes/BootLoaderUpdatesPhase1.

See:
- https://fedoraproject.org/wiki/Changes/BootLoaderUpdatesPhase1
- https://gitlab.com/fedora/bootc/tracker/-/issues/61
- https://gitlab.com/fedora/bootc/base-images/-/merge_requests/90

Draft PRs for GRUB & shim that will use the new paths:
- https://src.fedoraproject.org/rpms/grub2/pull-request/136
- https://src.fedoraproject.org/rpms/grub2/pull-request/140
- https://src.fedoraproject.org/rpms/shim/pull-request/4

travier, 2025-05-27:
Ah, there is no dist for shim as it's the same build for all releases

travier, 2025-05-27:
So this makes installing the shim & grub binaries under versioned directories less interesting.

lsandov1, 2025-05-27:
I have reworked MRs [1,2] (no scratch builds this time) based on your comments and the only piece different is new `efi` folder on top of `/usr/lib/`, where grub and shim will place their files on top

```
root@fedora rpms-unification]# rpm -ql grub2-efi-x64 shim-x64 | grep /usr/lib/efi
/usr/lib/efi/grub2/2.12-34.fc42/EFI/fedora/grubx64.efi
/usr/lib/efi/shim/15.8-4/EFI/BOOT/BOOTX64.EFI
/usr/lib/efi/shim/15.8-4/EFI/BOOT/fbx64.efi
/usr/lib/efi/shim/15.8-4/EFI/fedora/BOOTX64.CSV
/usr/lib/efi/shim/15.8-4/EFI/fedora/mmx64.efi
/usr/lib/efi/shim/15.8-4/EFI/fedora/shim.efi
/usr/lib/efi/shim/15.8-4/EFI/fedora/shimx64.efi
```

but maybe this does not solve the issue.

@travier @HuijingHei propose any directory hierarchy and I follow it, whatever if fits better for your scripting.

HuijingHei, 2025-07-21:
Copy @cgwalters 's comment here in case it is buried.

Yes per previous discussion it's a backwards incompatible change to use the new path - the old bootupd (before this PR) won't know what to do with data generated by bootupd after this PR.

Let's define some terms here:

## Layouts:

### /boot/efi/EFI

The default place for Fedora grub/shim RPM packages.

### /usr/lib/ostree-boot

Where rpm-ostree moves the above to by default today and what bootupd consumes.

### /usr/lib/bootupd/updates/EFI

This is content owned by bootupd.

### /usr/lib/efi/&lt;component&gt;/&lt;version&gt;

What the proposed new Fedora layout uses (and we could try to champion cross-OS/distro).

---

So for now we need to go:

/usr/lib/ostree-boot -> /usr/lib/bootupd/updates/EFI
/usr/lib/efi/&lt;component&gt; -> /usr/lib/bootupd/updates/EFI

But ideally in a future PR we teach bootupd to *also* know how to directly check for updates in /usr/lib/efi/&lt;component&gt; for EFI. We'll need to ship the code that knows how to consume those long before people can switch the default though.

HuijingHei, 2025-07-21:
> But ideally in a future PR we teach bootupd to _also_ know how to directly check for updates in /usr/lib/efi/<component> for EFI. We'll need to ship the code that knows how to consume those long before people can switch the default though.

What do you think of splitting like this? CC @cgwalters 
### Step 1
Copy /usr/lib/efi/<component> -> /usr/lib/bootupd/updates/EFI
This will make the installation / updates works as before.

### Step 2
Transfer `/usr/lib/ostree-boot -> /usr/lib/efi/<component>`
Copy `/usr/lib/efi/<component> -> /usr/lib/bootupd/updates/EFI` (step 1)
Change installation to read from `/usr/lib/efi/<component>`

### Step 3
Finally change the updates to read from `/usr/lib/efi/<component>`

cgwalters, 2025-07-25:
Yes, agreed! The "dual layout" approach should help unblock testing a lot.

HuijingHei, 2025-09-13:
> ### Step 1
> Copy /usr/lib/efi/ -> /usr/lib/bootupd/updates/EFI This will make the installation / updates works as before.
> 
> ### Step 2
> Transfer `/usr/lib/ostree-boot -> /usr/lib/efi/<component>` Copy `/usr/lib/efi/<component> -> /usr/lib/bootupd/updates/EFI` (step 1) Change installation to read from `/usr/lib/efi/<component>`
> 
> ### Step 3
> Finally change the updates to read from `/usr/lib/efi/<component>`

I did some testing if change installation to read from `/usr/lib/efi/<component>`, 
- It will make the update not working correctly, for example if change is `shim/15.8-5/EFI/fedora/shim.efi` (source file under `usr/lib/efi`), should change the [apply_diff()](https://github.com/coreos/bootupd/blob/main/src/filetree.rs#L223) to find the correct destination `fedora/shim.efi`.
- Also need to change [diff()](https://github.com/coreos/bootupd/blob/main/src/filetree.rs#L223) to get correct list when we update to new filetree.
So I will move the installation to Step3.

### Step 1
Copy `/usr/lib/efi/ -> /usr/lib/bootupd/updates/EFI`
 
### Step 2
Transfer `/usr/lib/ostree-boot -> /usr/lib/efi/<component>`

### Step 3
Change installation to read from `/usr/lib/efi/<component>`
Finally change the updates to read from `/usr/lib/efi/<component>`