coreos/bootupd #933: "Ignoring downgrade" with older grub2-efi installed

miabbott, 2025-05-21:
I upgraded my system to Fedora 42 (specifically a custom build of Silverblue 42 using the Universal Blue tooling), which includes `grub2-efi-ia32-2.12-28.fc42.x86_64` and `grub2-efi-x64-2.12-28.fc42.x86_64`, etc

When I check the output of `bootupctl status`, I see the following:

```
$ sudo bootupctl status 
Running as unit: bootupd.service
Component EFI
  Installed: grub2-efi-ia32-1:2.12-21.fc41.x86_64,grub2-efi-x64-1:2.12-21.fc41.x86_64,shim-ia32-15.8-3.x86_64,shim-x64-15.8-3.x86_64
  Update: Ignoring downgrade
No components are adoptable.
Boot method: EFI
```

It looks like the `shim` packages have successfully upgraded the EFI binaries, but the `grub2-efi` packages have not.

When I try `bootupctl update`, it reports no updates avaiilable.

```
$ sudo bootupctl update
Running as unit: bootupd.service
No update available for any component.
```

Nothing in the journal points to a failure or error.

cgwalters, 2025-06-30:
Parsing version numbers gets into the thorny issue of cross-distribution version strings. There's of course https://uapi-group.org/specifications/specs/version_format_specification/ and maybe we could try to constrain our inputs to match that.

cgwalters, 2025-08-08:
This is I think probably a relatively unusual case in that the build for Fedora N-1 is chronologically newer than the version for Fedora N.

However...instead of going all the way to comparing version numbers in the same way RPM does, we could simply incorporate the data from `/usr/lib/os-release` into our calculations. If the combination of either `$ID` or `$VERSION_ID` have changed, we assume that we need to apply the update, regardless of timestamp.

But *inside* an "$ID-$VERSION_ID" (an OS stream/release/version) I think it's pretty safe to just rely on timestamps and not tie ourselves to any specific package system comparison method.

cgwalters, 2025-08-13:
You are right that trying to adapt to the `/usr/lib/efi` format would force comparing version numbers.

However, one option here is to suggest that OSes using this conform to https://uapi-group.org/specifications/specs/version_format_specification/ instead.

One way to do that I think for Fedora would be to start by ignoring the epoch. If a component there ever needed to bump epoch, we'd have to instead map it to one of `~` or `^`.

But actually since nowadays RPM supports those too I think we could just hopefully not ever bump epoch again.