coreos/bootupd #934: Enabling AI code review

cgwalters, 2025-05-22:
Hi, I'd like to propose installing https://github.com/apps/gemini-code-assist on this repository. There's a lot of other options out there, but I've been impressed in my experience with it.

Of course it's a fully proprietary tooling and model, and I always feel tension in using proprietary code to develop FOSS.

Please 👍 on this issue if you're OK, and if you have feelings against, please say here. Otherwise if no one has objections I plan to enable this soon. We can always change things later of course.

HuijingHei, 2025-05-27:
I am OK with both, but I think it would make the work more efficiently in some cases, maybe we can add it in the future when the tool is more powerful.

travier, 2025-07-08:
This will happen with https://github.com/coreos/repo-templates/pull/304 via https://github.com/coreos/bootupd/pull/954.

closed 2025-07-24 by travier