coreos/bootupd #989: `generate-update-metadata` will miss shim files if upgrade to `grub2-2.12-41.fc44`

HuijingHei, 2025-08-25:
The goal is to have both updated grub2 & shim, see https://github.com/coreos/bootupd/issues/926.

But in the case that we only have grub2 update (https://bodhi.fedoraproject.org/updates/FEDORA-2025-e5c1f79161), and simply checking  https://github.com/coreos/bootupd/blob/cf75e14d55fd11b8f7bcad7a66fa359b9d10b963/src/efi.rs#L440, this will lead to shim files are missing, 
```
00:55:18  >>>> Executing `postprocess` inline script '8' <<<<
00:55:18  
00:55:18  + /usr/bin/bootupctl backend generate-update-metadata
00:55:18  Generated update layout for BIOS: grub2-tools-1:2.12-41.fc44.x86_64
00:55:18  Generated update layout for EFI: grub2-1:2.12-41.fc44
```
finally got failed:
```
00:59:39  org.osbuild.bootupd: 5c1b77fbf5bd6388637c2131f636946a93e53b48e34d1b8443ab1f89575d5521 {
00:59:39    "bios": {
00:59:39      "device": "disk"
00:59:39    },
00:59:39    "static-configs": true,
00:59:39    "deployment": {
00:59:39      "default": true
00:59:39    }
00:59:39  }
00:59:39  device/disk (org.osbuild.loopback): loop0 acquired (locked: False)
00:59:39  mount/root (org.osbuild.xfs): mounting /dev/loop0p4 -> /home/jenkins/agent/workspace/test-override/cache/osbuild/store/tmp/buildroot-tmp-ysu0_ygb/mounts/
00:59:40  mount/boot (org.osbuild.ext4): mounting /dev/loop0p3 -> /home/jenkins/agent/workspace/test-override/cache/osbuild/store/tmp/buildroot-tmp-ysu0_ygb/mounts/boot
00:59:40  mount/efi (org.osbuild.fat): mounting /dev/loop0p2 -> /home/jenkins/agent/workspace/test-override/cache/osbuild/store/tmp/buildroot-tmp-ysu0_ygb/mounts/boot/efi
00:59:40  Failed to open file "/sys/fs/selinux/checkreqprot": Read-only file system
00:59:40  error: boot data installation failed: Failed to find shimx64.efi in the image
00:59:40  Traceback (most recent call last):
00:59:40    File "/run/osbuild/bin/org.osbuild.bootupd", line 82, in <module>
00:59:40      r = main(_args, _args["options"])
00:59:40    File "/run/osbuild/bin/org.osbuild.bootupd", line 75, in main
00:59:40      subprocess.run(cmd, check=True)
00:59:40      ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
00:59:40    File "/usr/lib64/python3.13/subprocess.py", line 577, in run
00:59:40      raise CalledProcessError(retcode, process.args,
00:59:40                               output=stdout, stderr=stderr)
00:59:40  subprocess.CalledProcessError: Command '['chroot', '/run/osbuild/mounts/ostree/deploy/fedora-coreos/deploy/41980f98416921c2fa071affd7aa065fb1ecc24f69a48f6f4401bb8c600ab5db.0', '/usr/bin/bootupctl', 'backend', 'install', '--device=/dev/loop0', '--with-static-configs', '/run/osbuild/mounts']' returned non-zero exit status 1.
```

dustymabe, 2025-08-25:
If it makes it easier to just assume we have both, I'm OK with that. How long until the new update for shim comes out?

HuijingHei, 2025-08-26:
It is not easy to fall back, from the new grub2 file `/usr/lib/efi/grub2/1:2.12-41.fc44/EFI/fedora/grubx64.efi` and will not translated to `/usr/lib/ostree-boot`, I think that is why we only have `Generated update layout for EFI: shim-x64-15.8-3.x86_64`, but without grub2
```
rpm2cpio grub2-efi-x64-2.12-41.fc44.x86_64.rpm | cpio -idmv
./boot/grub2/fonts
./boot/grub2/fonts/unicode.pf2
./boot/grub2/grubenv
./boot/loader/entries
./etc/dnf/protected.d/grub2-efi-x64.conf
./etc/grub2-efi.cfg
./etc/grub2.cfg
./usr/bin/gen_grub_cfgstub
./usr/lib/efi/grub2/1:2.12-41.fc44/EFI/fedora/grubx64.efi
```
In this case, it is wired to partially copy `grubx64.efi` from `/usr/lib/efi`, and shim files from `/usr/lib/ostree-boot`, any better workaround for this? cc @travier

HuijingHei, 2025-08-26:
> In this case, it is wired to partially copy grubx64.efi from /usr/lib/efi, and shim files from /usr/lib/ostree-boot, any better workaround for this?

One easier workaround that I can think of is in postprocess before `generate-update-metadata`, copy `/usr/lib/efi/grub2/1:2.12-41.fc44/EFI/fedora/grubx64.efi` to `/usr/lib/ostree-boot/efi/`, then remove `/usr/lib/efi/`
see https://gitlab.com/fedora/bootc/base-images/-/merge_requests/278

HuijingHei, 2025-08-26:
I did testing with the above PR and it worked.
```
>>>> Executing `postprocess` inline script '8' <<<<

++ find /usr/lib/efi/ -name EFI
+ efi=/usr/lib/efi/grub2/1:2.12-41.fc44/EFI
++ echo /usr/lib/efi/grub2/1:2.12-41.fc44/EFI
++ wc -l
+ count=1
+ '[' 1 == 1 ']'
+ cp -r /usr/lib/efi/grub2/1:2.12-41.fc44/EFI /usr/lib/ostree-boot/efi/
+ rm -vrf /usr/lib/efi
removed '/usr/lib/efi/grub2/1:2.12-41.fc44/EFI/fedora/grubx64.efi'
removed directory '/usr/lib/efi/grub2/1:2.12-41.fc44/EFI/fedora'
removed directory '/usr/lib/efi/grub2/1:2.12-41.fc44/EFI'
removed directory '/usr/lib/efi/grub2/1:2.12-41.fc44'
removed directory '/usr/lib/efi/grub2'
removed directory '/usr/lib/efi'

>>>> Executing `postprocess` inline script '9' <<<<

+ /usr/bin/bootupctl backend generate-update-metadata
Generated update layout for BIOS: grub2-tools-1:2.12-41.fc44.x86_64
Generated update layout for EFI: grub2-efi-x64-1:2.12-41.fc44.x86_64,shim-x64-15.8-3.x86_64
```

closed 2025-08-29 by HuijingHei